StarLabsLtd/firmware #195: [StarLite Mk V] Very specific power button input needed to turn on device

edpenz, 2024-07-17:
The behaviour of the power button to turn on the device is a bit finicky and confused me for a long time:

* 1 short press -> device does not turn on
* 1 medium press -> light flashes once, around 7 seconds later light flashes a few times, device does not turn on
* 1 medium press, then again shortly after -> light flashes once on first press, light flashes a few times on second press, device turns on
* 1 long press -> light flashes a few times , device turns on

It took me a long time to figure out that the device can only reliably be turned on by making sure you hold the button for longer than a second or so. The first three behaviours are confusing though, as a medium press sometimes turns the device on, but only if you have recently failed to turn it on by a preceding medium press.

Tested on a device running firmware 24.06, Ubuntu 24.04.

Sean-StarLabs, 2024-07-17:
It's 200ms in G3 to target S5. 200ms from G3 to target S0.

Sean-StarLabs, 2024-07-19:
Tweaked in 102457fb747954f0f4b4481a8ff2e16cd9251e1d

closed 2024-07-19 by Sean-StarLabs