StarLabsLtd/firmware #196: [StarLite Mk V] make super key the actual super key and not menu?

BlackIkeEagle, 2024-07-18:
Is there a way to make the 'super' key on the detacheable keyboard actually sending the super key as on regular keyboards, now it is pretty annoying to have to remap the keys so everything works te same accros multiple machines.

Sean-StarLabs, 2024-07-18:
How do you mean? It sends `0xe0 0x1f`, same on all of our models

BlackIkeEagle, 2024-07-18:
Well I'm not comparing it to other StarLabs machines but on both a Tuxedo Laptop, a Dell Laptop and a Asus Laptop I get the following when I use `showkey` and I press the `windows` aka `super` key:

```
sudo showkey
kb mode was ?UNKNOWN?
[ if you are trying this under X, it might not work
since the X server is also reading /dev/console ]

press any key (program terminates 10s after last keypress)...
keycode 125 press
keycode 125 release
keycode 125 press
keycode 125 release
```

When I press the `super` key on the detacheable keyboard of the Starlite MK V I get:

```
sudo showkey
kb mode was ?UNKNOWN?
[ if you are trying this under X, it might not work
since the X server is also reading /dev/console ]

press any key (program terminates 10s after last keypress)...
keycode 127 press
keycode 127 release
keycode 127 press
keycode 127 release
```

So I'm in the wrong assumption it should do the same as the `windows` key I have on my other machines? If so it would be nice to have it configurable to it could behave the same for consistency across machines.

Sean-StarLabs, 2024-07-18:
Ah, that's older firmware for the keyboards - new firmware's going up to the LVFS so that'll change soon

BlackIkeEagle, 2024-07-18:
Does the Keyboard also need a `quirck` for `fwupd` for it to show up when I run `fwupdmgr get-devices`?

Sean-StarLabs, 2024-07-18:
It wont show up yet

Sean-StarLabs, 2024-10-22:
Current way to update: https://github.com/StarLabsLtd/starlite_keyboard_update

closed 2024-10-22 by Sean-StarLabs