StarLabsLtd/firmware #197: Starlite Mk V: Charging Status Toggles on And Off

thebatdan, 2024-07-19:
While connected to the charger, the Starlite will intermittently stop and start charging. It usually toggles from charging, to not charging, back to charging, within a second or two. This toggling happens multiple times in succession, and happens on either USB-C port.

The charger and cable seem to be ok, as I do not get this behavior charging other devices.

radekholy24, 2024-07-19:
I think this happens on my StarBook Mk VI too. Haven't tried to diagnose it yet but several times per day I can hear the OS's sound assigned to plugging and unplugging the charger with the same frequency as the OP. Feel free to ignore my comment, I just wanted to let you know that it might affect StarBooks too.

Sean-StarLabs, 2024-07-19:
Regression in 24.07.10, already fixed - will push build later.

No known issues on the StarBook

Sean-StarLabs, 2024-07-19:
Fixed in 102457fb747954f0f4b4481a8ff2e16cd9251e1d

dchang0, 2024-09-20:
Please re-open this ticket.

I am experiencing this exact same bug now in BIOS 24.08 and EC firmware 24.08. It was also happening in 24.07.

I am using the factory charger and factory USB C cable.

It seems to happen more frequently in the middle range between 20% battery and 80% battery. Once it gets to around 80% battery the frequent charging/not-charging flip-flops slow down and then disappear.

It does not appear to have anything to do with the OS. The red charging LED actually turns off when charging stops, then turns on when it starts.

baarkerlounger, 2024-09-23:
This also happens on the StarFigher (firmware 0.18) when the battery level is below 20%.

Sean-StarLabs, 2024-09-23:
> This also happens on the StarFigher (firmware 0.18) when the battery level is below 20%.

That would be expected as there's no hybrid power when the battery is at that level.

baarkerlounger, 2024-09-23:
Could you explain what you mean by hybrid power? I would have expected it to charge steadily regardless of power level.

Sean-StarLabs, 2024-09-23:
Hybrid power is where the battery can support the charger, if the laptop needs more power. That's disabled under 20%.

So under 20%, if you use all the power from the charger, then it'll keep the laptop running and stop charging the battery until there's some power spare (rather than limiting performance).

baarkerlounger, 2024-09-23:
I see, thank you. I expected that the charger would be able to provide the max power the laptop can draw + some to charge the battery, but I guess that's not the case then.

dchang0, 2024-09-24:
It MIGHT be that the charging fluctuates in sync with spikes in WiFi traffic. I kept the System Monitor running to see if it might be synced with CPU use, but it actually seems to be synced with spikes in the Network graph (I only have WiFi, no Ethernet).

I'll need to watch it more carefully every day to be sure it really is correlated to Network graph spikes.

closed 2024-10-17 by Sean-StarLabs

DevenBL, 2024-11-05:
It's still not fixed in 24.10. RIP

dchang0, 2024-11-05:
> It's still not fixed in 24.10. RIP

Same here.

I'm on 24.09 on the StarLite Mk. V and still see it happening too, but it is a lot less frequent than under 24.08. It tends to happen around the 80% +/-5% battery level while I'm doing stuff like accessing the network. Previously under 24.08, it happened all the way from 50% to 85% battery level.

Under 24.09, if I just let the tablet sit there, untouched, it seems to happen much less frequently than in 24.08, though it does still occasionally happen.

Until this is totally fixed, I really wish it were possible to turn off the sound effect it makes in Ubuntu when charging kicks in or out. I'll be watching a movie with the audio turned up through the Bluetooth speakers, and ba-doomp--pause--ba-doomp, the sound effect will blast through the speakers. But the sound is important too so that I know when it's charging, so it might be better to be able to disable or enable the sound on the fly, in the Settings panel.

Sean-StarLabs, 2024-11-05:
What settings?

dchang0, 2024-11-05:
> What settings?

I'm assuming you are directing this question towards me. When I refer to Settings, I mean the Ubuntu System Settings app, under the Power panel. Adding a checkbox there: "[ ] notification sound when charging" would be great.
It would be even better to also have a toggle under the quick-access menu (when you click the battery level icon) to the right of Airplane Mode.

If you are asking "what settings do you have in BIOS while the bug is still occurring?" I use all the factory default settings except for setting the charging speed to 0.5C (down from 1.0C) and turning off the keyboard lid switch.

Sean-StarLabs, 2024-11-05:
Thanks, worth trying 24.10 when that's released. Will re-open if that doesn't do it.

DevenBL, 2024-11-07:
It seems in addition to the above problems that reported charging wattage is still incorrect on 24.10 release at a 1C charging rate.
the wattage measured externally on the the power cable oscillates between 0 and 60 watts while gradually discharging whereas the charging rate reported in the OS follows this general pattern. 

external watts: 0,   60, 60, 0, 0    , 0     0,
internal  watts: 0, 200, 15, 0, 200, 15, 0,
it then repeats on a synchronized cycle.

i am using the powersupply flatpack to report internal watts.

So yeah issue not fixed needs to get reopened.

Sean-StarLabs, 2024-11-07:
> still incorrect on 24.10 release

>> worth trying 24.10 when that's released

dchang0, 2024-11-07:
Thanks. Last night I got hit by a flurry of rapid disconnect-reconnects while watching a video using VLC (but I use VLC all the time, and the disconnect-reconnects don't usually go nuts like this). It was around 75% to 80% and then stopped happening around 85%.

It feels like it's an oscillation/feedback problem similar to what happens with audio amplifier systems. Under specific circumstances, it happens and then somehow corrects itself. That's gotta be tough figuring out...

Sean-StarLabs, 2024-11-07:
It's an i2c operation timing out.

alexreid, 2024-11-27:
This is still happening for me on my Starlite Mk V running firmware 24.10 (using Linux kernel 6.11.8 in case that's relevant).

DevenBL, 2024-12-03:
This issue is still happening in some form on the 24.10 official release (through lvfs). The charging wattage is still misreported in the system. charging still oscillates although reduced. now between 100% and 98% instead of 95% and (eventually) 0% (as is the case for certain firmware versions).
charging reported at 20W is when externally measured actually at 35-45W. Occasionally internal charging reports astronomical figures of 200-300W when external measurement reports no anomalies. External measurements were done at the USB port. Charging this version is relatively stable in comparison but still occasionally disconnects at full charge or otherwise.
Astronomical charging figures are most commonly seen after charging disconnects and then resumes.

Good luck Sean! This version actually charges more than it discharges and actually can reach 100% charge. So definite progress!

Sean-StarLabs, 2024-12-03:
> external measurement 

Of what? Are you sure you're not just comparing charging current with input current, because they'll never be the same.

> The charging wattage is still misreported in the system.

Assuming it's not just the RBP, can you elaborate?

DevenBL, 2024-12-03:
> Are you sure you're not just comparing charging current with input current

If i understand what you mean correctly this is exactly what i am doing. I measure the charging wattage going into the tablet through the Usb port with additional non tablet external physical hardware (external measurement). And i am comparing that to the internal charging wattage being what the tablet reports through software as it's charging wattage.

to be clear i am using an external usb based power measurement device to measure the wattage between the wall charger and the tablet. And comparing that to what the tablet reports through software.

> because they'll never be the same.

Yes and i do not expect them to be exactly the same but they should be somewhere in line with each other. But if only 14 watts of power are going into the tablet from the outside world and the tablet reports charging at 20 watts (_and is actually charging IE: battery percentage going up_) something is wrong **_OR_** you should win a nobel prize for your discovery on how to violate the laws of thermodynamics by generating 6 watts out of thin air.

Additionally on the other end If the tablet is reporting charging at 300 watts (while externally measured power is measured at 40 watts) and my dinky Usb cable is not on fire, melting, hot or even warm. Clearly the charging wattage should be being misreported in some way shape or form in the other direction as well.

> Assuming it's not just the RBP, can you elaborate?

What does RBP mean here?

Software used to get the software reported internal charging wattage is btop or the flatpak: https://flathub.org/apps/nl.brixit.powersupply

Sean-StarLabs, 2024-12-03:
>  measure the wattage

Hmm, there are lots of unknowns here—the measurement could be from the POD, shunt, or other form of measurement, so there is no clue as to how accurate it is (especially as 14W is too low even with PPS - your original 35W in, 20W charging would be reasonable though). Most of the little USB-C measures are real-time;  you're comparing against averaged data that we don't know how recent it is as Q Events are only generated when:
* the charger is connected or disconnected
* the battery charge level changes by 1% or more
* the current changed by more than 2W

Duration of the measurements would be relevant too, plus, what's "happening" i.e. did you just  plug in the charger, as we'd expect to see 200W+ for the first reading (theoretically impossible to see over 255 though).

ACOK going down around 80% we can look into, but to look into rest, we're going to need some output from the [EC console](https://github.com/StarLabsLtd/merlin), as that spits out a message every time the input current, current and voltage changes—and that's read directly from the charge controller.

> What does RBP mean here?

Ring Buffer - simply put, averaged

>  Software used to get the software reported internal charging wattage is btop or the flatpak:

The common tool is upower. It just looks at the values in ACPI, which come straight from the EC (or from the EC console for the most accurate data).

DevenBL, 2024-12-04:
> Hmm, there are lots of unknowns here—the measurement could be from the POD, shunt, or other form of measurement, so there is no clue as to how accurate it is (especially as 14W is too low even with [...]

External measurement are accurate within +-1.5 watts at the very least and i can get it to charge at 14 watts for many minutes to perhaps hours with the internal reported wattage being at 20 watts for that period.
I checked and the wattage figures i have have posting for internal are consistent with what upower reports 
as for the averaging period for external measurement. I have no idea!

> ACOK going down around 80% we can look into

what does this mean?

> we're going to need some output from the [EC console](https://github.com/StarLabsLtd/merlin)

How do i obtain this? Do i need to compile and install merlin? Is there any simpler solution like cat a file or minicom or something?

> did you just plug in the charger, as we'd expect to see 200W+ for the first reading 

yes and no it seems to report 200+ watts only after it does the charging status toggles on and off thing that was originally reported. Unplug and replug of the charger in normal circumstance does not seem to do that. It generally seems to happen for the first few readings after charging resumes. It's generally inconsistent so i cant really say anything except it does it occasionally and normally after the charging stops for some reason.

> (theoretically impossible to see over 255 though)

I have not looked too hard at the peak power figure ive seen exactly but it's between 220-300W at least.

Additionally the upower reported wattage between 100% and 98% is highly unstable varying from at least 0W to at least 200W

Sean-StarLabs, 2024-12-04:
> i can get it to charge at 14 watts for many minutes to perhaps hours 

What charger, cable and measurer?

> what does this mean?

AC power "stopping"

>  How do i obtain this?

It's a link - https://github.com/StarLabsLtd/merlin, just clone, cd util/console, make, sudo ./merlin-console -c or sudo ./merlin-console -p (p is continuous)

> happen for the first few readings after charging resumes.

Okay, so that's expected

DevenBL, 2024-12-04:
> What charger, cable and measurer?

https://baseusonline.com/product/950/baseus-display-fast-charging-data-cable-type-c-to-type-c-100w-1m-black 
this model. Charger is from StarLabs. I stick to this cable mostly for convenience. Measurements are consistent with these https://www.startechtel.com/products/p3-international-p4400-kill-a-watt-electric-usage-p3ip4400 and a multi meter with a flayed usb cable or measurement at the wall with a multimeter.

DevenBL, 2024-12-08:
> ACOK going down around 80% we can look into, but to look into rest, we're going to need some output from the [EC console](https://github.com/StarLabsLtd/merlin), as that spits out a message every [...]

Here is the output of the EC console you asked for (./merlin-console -cp > output.txt) 
[output.txt](https://github.com/user-attachments/files/18053638/output.txt)

thebatdan, 2024-12-17:
Confirming that the issue is still happening for me as well on Firmware 24.10. 

Can we reopen this until the issue is resolved?

DevenBL, 2024-12-18:
Confirming that the discharging issue still occurs intermittently on 24.10 at 1C charge rate. Although certainly less often than past builds.

thebatdan, 2025-01-31:
I've updated to firmware 25.01 from LVFS. Still no change to this issue, charging still randomly toggles on and off. Couple of things I've noticed around this issue:
* I tested with a different, 90w charger. The issue does not happen with that charger.
* Dropping the charging rate down to 0.2C in the bios results in an excruciatingly long charge time, and the issue still occurs at about 90% charge.
* I was using the starlite on a different laptop, and had my phone charging of the starlite charger USB port as well. This made the issue appear on the other laptop as well, but it stopped happening once I removed the phone from charging.

Is this an issue with the Starlite, or with the charger?

Can this issue be reopened until it is actually resolved, @Sean-StarLabs ?

dchang0, 2025-02-01:
I'm not 100% sure this is related to firmware 25.01, but right after updating to it, I've had this problem where if I suspend the tablet for several minutes (long enough to walk a mile from the coffee shop), when I get back, I can wake the tablet from suspend, but I can't log in to Ubuntu 24.04 (factory installed OS). Specifically, I can't click into the password field to start typing.

It has happened several times now; I have to press and hold the power button to force a reboot. The Ubuntu 24.04 GUI login screen is responsive otherwise; I can move the mouse pointer and interact with the system menu at the top right corner. I can put it into suspend and wake it repeatedly, but it will never let me type the password to log in.

I'd have to downgrade the firmware back to 24.10 to be sure it is caused by it.
I'm also looking for any combination of keys to press on the keyboard cover to put focus on the password input field.

Anyone else seeing this problem?

Sean-StarLabs, 2025-02-01:
@thebatdan Whats the output of `cat /sys/class/dmi/id/ec_firmware_release`?

>  was using the starlite on a different laptop, and had my phone charging of the starlite charger USB port as well. This made the issue appear on the other laptop as well, but it stopped happening once I removed the phone from charging.

Yes, you have less power available so there will be a deficit

@dchang0 It's Ubuntu - they released some updates 2/3 days ago that seem to fix it.

thebatdan, 2025-02-02:
@Sean-StarLabs: 
```
cat /sys/class/dmi/id/ec_firmware_release
25.1
```

Regarding the phone charging, I'm fine with not being able to charge both laptop and phone at the same time. Just made me wonder if the starlite is maybe drawing more power than the charger is capable of delivering.

thebatdan, 2025-02-14:
@Sean-StarLabs any update on this issue?

thebatdan, 2025-02-23:
@Sean-StarLabs what's the update here? When can we expect to be able to charge the StarLite with the actual charger that it came with?

Can we reopen this issue until it's resolved?

dchang0, 2025-02-23:
I agree. We should switch to a "broken until proven working" attitude here, keep the ticket open as long as it takes, and focus on gathering as much feedback from end users in real-world conditions to put the problem to bed once and for all.

Sean-StarLabs, 2025-02-23:
New build should hit stable Tuesday