StarLabsLtd/firmware #200: [StarLite Mk V] Need to hold power button to wake from S3-deep/suspend-to-ram state

rbm78bln, 2024-07-22:
Hi there,

I'm not fully sure, whether is was implemented intentionally or not, but when trying to wake the StarLite Mk V from suspend to ram state (s3/deep) when the button is slowly pulsing, you need to press and hold the button for 2-3 secs to actually make it come back to life again. On the other hand it takes just a short press of the button to turn it back on from S5 state and also from S2-idle. Also I don't know whether it was a deliberate decision, to keep the power button fully lit while the device is in S2-idle state. 

For my personal taste, it would be nice to have the same wake-up behaviour of the power button for waking up from S5, S3-str, and S2-idle by just a short press of the button as it is done for S5 and idle now. Also it would be nice to be able to differenciate by the power led, whether the device is in S0-working or S2-idle mode. Nevertheless I do like that I am able to tell S2-idle and S3-str  apart from looking at the led.

Thanks guys!

Sean-StarLabs, 2024-07-22:
Yeah, 2 second delay on the power button when in S0; already changed -- https://github.com/StarLabsLtd/firmware/issues/184

Power LED on means it's in S0 - we don't support MODS.

Sean-StarLabs, 2024-10-15:
Modern Standy (S2-idle is part of that)