StarLabsLtd/firmware #203: [StarLite Mk V] TPM is not usable

chimpanzee23, 2024-07-26:
I would like to use the TPM in the StarLite Mk V for measured boot and automatic LUKS decryption.

Running Aeon Desktop, the TPM is not recognised by the OS using the default BIOS settings. Enabling Intel ME allows the OS to recognise the TPM (I assume Intel PTT), but the systemd-abrmd service fails to start and tpm2_selftest fails.

s3ph-scott, 2024-07-30:
I am seeing a similar situation with Ubuntu 24.04. I have tried multiple versions of Coreboot (24.05, 24.06 and 24.07) with both Intel Management Engine enabled and disabled. In all instances the experimental option for TPM encryption is disabled by the Ubuntu installer.

To try and understand what's happening I loaded an install image for Windows 11. The installer did not block me from starting the installation but due to not having a compatible driver available on the install media I was unable to progress past the initial prompt to load a device driver for Windows 11. That is to say that this would be inconclusive as I don't know at which stage a Windows 11 install verifies the availability of a TPM and Secure Boot.

The [specification page](https://starlabs.systems/pages/starlite-specification) for the Starlite MKV does not mention a separate TPM chip although on the [overview page](https://starlabs.systems/pages/starlite) there is a mockup of an Infineon Optiga chip. I would have to guess on the Starlite MKV the TPM is firmware based and from my limited knowledge it may depend on Intel ME being enabled.

Sean-StarLabs, 2024-07-30:
edk2's implementation of TPM support with coreboot doesn't work with FDE; it's on our radar, but there aren't any timescales at the moment it hasn't been established at which end the problem lies.

s3ph-scott, 2024-07-30:
Thank you @Sean-StarLabs for the explanation. I will be using LUKS to satisfy device encryption in place of TPM backed encryption with a view to revisiting as and when the situation changes.

chimpanzee23, 2024-07-30:
Hi @Sean-StarLabs, thank you for the update, good to know it's on your radar. Are you happy for me to leave the issue open to track this and information for anyone else who has the same question?

Also, are you able to confirm whether the StarLite Mk V has a discrete TPM or uses Intel PTT?

Sean-StarLabs, 2024-07-30:
Of course. Its PTT

Sean-StarLabs, 2024-09-06:
Pretty sure it's fixed with 00bd1d2ebfecc59f98d72892159bdee07ae12d36 . Haven't tested all the possible uses - feel free to re-open if not.

chimpanzee23, 2024-09-06:
Hi @Sean-StarLabs, I've tested firmware 24.08 with the Intel ME enabled on both openSUSE Aeon and the Ubuntu 24.04.1 live iso and unfortunately the TPM still doesn't seem to be available. My dmesg output on both shows the below:

`
[    0.433828] [      T1] tpm_tis MSFT0101:00: [Firmware Bug]: failed to get TPM2 ACPI table
[    0.433849] [      T1] tpm_tis MSFT0101:00: probe with driver tpm_tis failed with error -22
[    0.433866] [      T1] tpm_crb MSFT0101:00: [Firmware Bug]: failed to get TPM2 ACPI table
[    0.433867] [      T1] tpm_crb MSFT0101:00: probe with driver tpm_crb failed with error -22
[    0.891559] [      T1] ima: No TPM chip found, activating TPM-bypass!
`

jothgard, 2024-09-09:
same here! Or do i need to reinstall after updating to 24.08?

dmesg |grep -i tpm: 
[    0.839899] ima: No TPM chip found, activating TPM-bypass!
[   16.157683] systemd[1]: systemd 255.4-1ubuntu8.4 running in system mode (+PAM +AUDIT +SELINUX +APPARMOR +IMA +SMACK +SECCOMP +GCRYPT -GNUTLS +OPENSSL +ACL +BLKID +CURL +ELFUTILS +FIDO2 +IDN2 -IDN +IPTC +KMOD +LIBCRYPTSETUP +LIBFDISK +PCRE2 -PWQUALITY +P11KIT +QRENCODE +TPM2 +BZIP2 +LZ4 +XZ +ZLIB +ZSTD -BPF_FRAMEWORK -XKBCOMMON +UTMP +SYSVINIT default-hierarchy=unified)
[   16.504146] systemd[1]: systemd-pcrextend.socket - TPM2 PCR Extension (Varlink) was skipped because of an unmet condition check (ConditionSecurity=measured-uki).
[   16.538755] systemd[1]: systemd-pcrmachine.service - TPM2 PCR Machine ID Measurement was skipped because of an unmet condition check (ConditionSecurity=measured-uki).
[   16.541227] systemd[1]: systemd-tpm2-setup-early.service - TPM2 SRK Setup (Early) was skipped because of an unmet condition check (ConditionSecurity=measured-uki).

chimpanzee23, 2024-09-09:
Hi @jothgard, your dmesg output doesn't have the tpm_tis and tpm_crb lines as mine does, do you have the Intel ME disabled? The TPM is PTT so will need the ME enabled once Sean has ironed out the kinks. I also had the systemd lines but omitted them as they're not too relevant to detecting the TPM.

I'm by no means expert but I would expect the TPM to just be usable once the firmware is fixed without re-installing, but I'm not sure if you can enrol it to decrypt your drive after installation on Ubuntu, I think I remember reading somewhere that Ubuntu gets it's bootloader and/or kernel from snap when you select the TPM option at install.

chimpanzee23, 2024-09-14:
I'll state again I'm not an expert in this, I'm just another Starlite user, but Sean confirmed earlier in this issue that the TPM is firmware-based Intel PTT. My understanding (from Google) is that PTT requires part of the ME so I think it will need to be enabled to use the TPM.

Sean-StarLabs, 2024-10-17:
Fixed with d6de2690dd547a014309086a9cc435e2b58f091e

closed 2024-10-17 by Sean-StarLabs

chimpanzee23, 2024-10-18:
Hi @Sean-StarLabs , I have installed firmware 24.09 on my Starlite and tested with Aeon Desktop, Ubuntu 24.04.1 and Ubuntu 24.10, but unfortunately I am getting exactly the same errors as last time in [24.08](https://github.com/StarLabsLtd/firmware/issues/203#issuecomment-2334711183). I have tried this with Intel ME both enabled and disabled.

Sean-StarLabs, 2024-10-18:
ME definitely active?

chimpanzee23, 2024-10-18:
I enabled it in the coreboot settings. Is there any way to check its actually enabled in the OS?

PyroDevil, 2024-10-20:
I also have issues getting TPM and Secureboot to work.

I updated to `24.09`, enabled the ME in bios. I tried enabling secure boot, but some options are disabled. I tried setting the mode to custom from standard, but that setting doesn't seem to stick, exiting the secure boot menu and entering it again resets the mode to standard.

Kernel log about TPM:

```
❯ sudo dmesg | grep -i tpm
[    1.436829] tpm_tis MSFT0101:00: [Firmware Bug]: failed to get TPM2 ACPI table
[    1.436850] tpm_tis MSFT0101:00: probe with driver tpm_tis failed with error -22
[    1.436868] tpm_crb MSFT0101:00: [Firmware Bug]: failed to get TPM2 ACPI table
[    1.436869] tpm_crb MSFT0101:00: probe with driver tpm_crb failed with error -22
[    1.977675] ima: No TPM chip found, activating TPM-bypass!
[    2.203129] systemd[1]: systemd 255.13-1.fc40 running in system mode (+PAM +AUDIT +SELINUX -APPARMOR +IMA +SMACK +SECCOMP -GCRYPT +GNUTLS +OPENSSL +ACL +BLKID +CURL +ELFUTILS +FIDO2 +IDN2 -IDN -IPTC +KMOD +LIBCRYPTSETUP +LIBFDISK +PCRE2 +PWQUALITY +P11KIT +QRENCODE +TPM2 +BZIP2 +LZ4 +XZ +ZLIB +ZSTD +BPF_FRAMEWORK +XKBCOMMON +UTMP +SYSVINIT default-hierarchy=unified)
[    2.292291] systemd[1]: systemd-pcrphase-initrd.service - TPM2 PCR Barrier (initrd) was skipped because of an unmet condition check (ConditionSecurity=measured-uki).
[   15.171090] systemd[1]: systemd 255.13-1.fc40 running in system mode (+PAM +AUDIT +SELINUX -APPARMOR +IMA +SMACK +SECCOMP -GCRYPT +GNUTLS +OPENSSL +ACL +BLKID +CURL +ELFUTILS +FIDO2 +IDN2 -IDN -IPTC +KMOD +LIBCRYPTSETUP +LIBFDISK +PCRE2 +PWQUALITY +P11KIT +QRENCODE +TPM2 +BZIP2 +LZ4 +XZ +ZLIB +ZSTD +BPF_FRAMEWORK +XKBCOMMON +UTMP +SYSVINIT default-hierarchy=unified)
[   15.605615] systemd[1]: systemd-pcrextend.socket - TPM2 PCR Extension (Varlink) was skipped because of an unmet condition check (ConditionSecurity=measured-uki).
[   15.661527] systemd[1]: systemd-pcrmachine.service - TPM2 PCR Machine ID Measurement was skipped because of an unmet condition check (ConditionSecurity=measured-uki).
[   15.664547] systemd[1]: systemd-tpm2-setup-early.service - TPM2 SRK Setup (Early) was skipped because of an unmet condition check (ConditionSecurity=measured-uki).
```

About secureboot:

```
❯ sudo dmesg | grep -i secureboot
[    0.000000] secureboot: Secure boot disabled
[    0.004844] secureboot: Secure boot disabled
```

Used system: Fedora 40 Silverblue, Linux 6.11.3-200.fc40.x86_64

```
Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: coreboot
	Version: 24.09
	Release Date: 10/17/2024
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		PC Card (PCMCIA) is supported
		BIOS is upgradeable
		Selectable boot is supported
		ACPI is supported
		Targeted content distribution is supported
	BIOS Revision: 24.8
	Firmware Revision: 24.7

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: Star Labs
	Product Name: StarLite
	Version: 1.0
	Serial Number: I500000
	UUID: Not Settable
	Wake-up Type: Reserved
	SKU Number: I5
	Family: I5
```

Sean-StarLabs, 2024-10-21:
> I enabled it in the coreboot settings. Is there any way to check its actually enabled in the OS?

`fwupdmgr security --force` is the most accessible.

PyroDevil, 2024-10-21:
This is what I got:

```
❯ sudo fwupdmgr security --force
Host Security ID: HSI:0! (v1.9.26)

HSI-1
✔ MEI key manifest:              Valid
✔ csme manufacturing mode:       Locked
✔ csme override:                 Locked
✔ csme v0:16.50.0.1120:          Valid
✔ Platform debugging:            Disabled
✔ Supported CPU:                 Valid
✔ UEFI bootservice variables:    Locked
✘ BIOS firmware updates:         Disabled
✘ SPI write:                     Not found
✘ SPI lock:                      Not found
✘ SPI BIOS region:               Not found
✘ TPM v2.0:                      Not found
✘ UEFI secure boot:              Not found

HSI-2
✔ Intel BootGuard:               Enabled
✔ Platform debugging:            Locked
✘ Intel BootGuard ACM protected: Invalid
✘ Intel BootGuard OTP fuse:      Invalid
✘ Intel BootGuard verified boot: Invalid
✘ IOMMU:                         Not found

HSI-3
✔ CET Platform:                  Supported
✔ Pre-boot DMA protection:       Enabled
✘ Intel BootGuard error policy:  Invalid
✘ Suspend-to-idle:               Disabled
✘ Suspend-to-ram:                Enabled

HSI-4
✔ SMAP:                          Enabled
✘ Encrypted RAM:                 Not supported

Runtime Suffix -!
✔ fwupd plugins:                 Untainted
✔ CET OS Support:                Supported
✔ Linux swap:                    Encrypted
✔ Linux kernel:                  Untainted
✘ Linux kernel lockdown:         Disabled

This system has a low HSI security level.
 » https://fwupd.github.io/hsi.html#low-security-level

This system has HSI runtime issues.
 » https://fwupd.github.io/hsi.html#hsi-runtime-suffix

Host Security Events
  2024-10-21 11:03:20:  ✔ CET OS Support changed: Not supported → Supported
  2024-10-21 11:00:50:  ✘ CET OS Support changed: Supported → Not supported
  2024-10-21 10:59:27:  ✔ CET OS Support changed: Not supported → Supported
  2024-10-20 12:13:27:  ✘ CET OS Support changed: Supported → Not supported
  2024-10-19 13:41:52:  ✔ Intel BootGuard changed: Not supported → Enabled
  2024-08-21 15:26:51:  ✘ Intel BootGuard changed: Not found → Not supported
  2024-07-18 12:25:21:  ✘ SPI write changed: Enabled → Not found
  2024-07-18 12:25:21:  ✘ SPI lock changed: Enabled → Not found
  2024-07-18 12:25:21:  ✘ SPI BIOS region changed: Unlocked → Not found
  2024-07-18 10:44:01:  ✔ CET OS Support changed: Not supported → Supported
```

chimpanzee23, 2024-10-21:
This one is mine, looks the same as @PyroDevil 
```
#Host Security ID: HSI:0! (v1.9.25)

HSI-1
✔ MEI key manifest:              Valid
✔ csme manufacturing mode:       Locked
✔ csme override:                 Locked
✔ csme v0:16.50.0.1120:          Valid
✔ Platform debugging:            Disabled
✔ Supported CPU:                 Valid
✔ UEFI bootservice variables:    Locked
✘ BIOS firmware updates:         Disabled
✘ SPI write:                     Not found
✘ SPI lock:                      Not found
✘ SPI BIOS region:               Not found
✘ TPM v2.0:                      Not found
✘ UEFI secure boot:              Not found

HSI-2
✔ Intel BootGuard:               Enabled
✔ Platform debugging:            Locked
✘ Intel BootGuard ACM protected: Invalid
✘ Intel BootGuard OTP fuse:      Invalid
✘ Intel BootGuard verified boot: Invalid
✘ IOMMU:                         Not found

HSI-3
✔ CET Platform:                  Supported
✔ Pre-boot DMA protection:       Enabled
✘ Intel BootGuard error policy:  Invalid
✘ Suspend-to-idle:               Disabled
✘ Suspend-to-ram:                Enabled

HSI-4
✔ SMAP:                          Enabled
✘ Encrypted RAM:                 Not supported

Runtime Suffix -!
✔ fwupd plugins:                 Untainted
✔ Linux swap:                    Encrypted
✔ Linux kernel:                  Untainted
✘ CET OS Support:                Not supported
✘ Linux kernel lockdown:         Disabled
```

Sean-StarLabs, 2024-10-22:
Thanks - can you try this? Download, extract, `sudo flashrom -p internal -w coreboot.rom -i bios --ifd -n -N` then shutdown.

[coreboot.zip](https://github.com/user-attachments/files/17475495/coreboot.zip)

chimpanzee23, 2024-10-22:
Hi Sean, I've tested the new coreboot.rom you supplied and the TPM is now recognised by the OS. Unfortunately, it only seems to be measuring PCR10 so is still unable to be used to decrypt the drive. This seems to be the case in both Aeon and Ubuntu. Seems like a step in the right direction though.

Sean-StarLabs, 2025-10-08:
Fixed with 703776874eff8fb24ed4bf828c09b7cf01cf9d1a