StarLabsLtd/firmware #207: [StarLite Mk V] Enable/Support Secure Boot

meeuw, 2024-08-15:
It would be nice to have support for Secure Boot for the StarLite Mk V

Sean-StarLabs, 2024-09-02:
Duplicate of #71

closed 2024-09-02 by Sean-StarLabs

stikonas, 2024-11-27:
Seems to work on Starbook MkVI (with own generated keys) though a bit pointless without BIOS password as anybody can just turn it off. I guess even with BIOS password one can do that if they externally reflash the firmware but that's is a bit more involved.

Sean-StarLabs, 2024-11-29:
Yeah, the general view seems to be that if someone gets hold of the computer, then you've "lost" and need to take some form of action to mitigate that.

The only technology that can protect a computer if someone gets hold of it is BootGuard. Intel's term is FPF (Field Programmable Fuses), but it basically burns something into the processor, which means the firmware must match a key in the firmware; otherwise, it won't start.  However, that has caveats:
* You need a hardware programmer and an NDA with Intel (which anyone can get) to active it
* The Intel ME must be on
* You need to build your own firmware and sign it with said key
* If the key is compromised, you can't change it, so the whole computer is compromised
* If the key is lost, there is nothing that can recover or change it - the only real option is to buy a new one