StarLabsLtd/firmware #208: Starlite Mk V Ubuntu battery charge indicator stays at 100%, charging LED always red

dchang0, 2024-08-31:
The battery charge level icon in the factory-installed Ubuntu always says 100% and the charge LED always stays red since yesterday.

I suspect it was caused by this somewhat common issue related to the keyboard cover waking the tablet from suspend: I put the tablet into suspend (usually by closing the cover, but this time by clicking Power > Suspend. I then put the tablet in my bag and carry it home from the coffee shop. The keyboard's keys are depressed and wake the tablet. Most of the time, the tablet stays awake the whole time (getting hotter) until I get home. This time, though, it went into some weird unresponsive state where the power light was rapidly flashing white. I had to hold down the power button for a long time to force it to turn on. This exact sequence has only happened once before (usually the tablet simply stays on once woken).

The first time this sequence happened, the battery charge level was not affected.

How do I forcibly reset the power management so that it properly displays the current charge level and charging state?

Sean-StarLabs, 2024-08-31:
> How do I forcibly reset the power management so that it properly displays the current charge level and charging state?

Turn it off, disconnect the charger and wait 12 seconds.

dchang0, 2024-08-31:
Thanks, Sean-- I confirm that this short procedure did work. For a second (before I tried it, the battery level indicator said 0% (when it obviously had more than 0% charge). I tried the procedure, then it said 98% and appears to be working properly.

closed 2024-08-31 by dchang0