StarLabsLtd/firmware #210: [StarLite Mk V] Sometimes crashing on lid close

PhilT, 2024-09-05:
I've noticed that sometimes when I close the lid the tablet starts flashing the power led very rapidly. I can then only turn it back on by holding the power button for about 10 seconds.

I'm not using the device a great deal at the moment so I'm not sure if it's something in particular that's running but I'm normally running several terminals and Firefox on NixOS.
 
On a more general note. Are the outstanding issues with this device being worked on? It's been a little while since it was released and I feel it could do with a bit more polish.

Thanks.

Sean-StarLabs, 2024-09-05:
Not known, and would need version numbers to look into.

> Are the outstanding issues with this device being worked on?

Yes

PhilT, 2024-09-05:
Version numbers?

Sean-StarLabs, 2024-09-05:
Kernel and firmware

PhilT, 2024-09-05:
Also, sometimes it doesn't suspend at all. Screen goes blank (as far as I can tell) but power LED stays on.

Sean-StarLabs, 2024-09-05:
Might be #183, can you trying with the tray out?

PhilT, 2024-09-05:
With the tray out it seems to sleep pretty quickly but Firefox still crashes and closing the lid is still a bit hit and miss.

Sean-StarLabs, 2024-09-05:
The only other known thing floating around is the 6.9 kernel, so worth trying 6.8, but I'd imagine it's something software related - tried another distro?

PhilT, 2024-09-05:
> tried another distro?
I haven't but I was contemplating it. I'll give that a try when I get the chance.
I'll also check 6.8 as well.
Thanks.

dchang0, 2024-09-06:
I've seen the same thing as the OP occasionally. I have also seen the same extremely rapid flashing power LED sometimes when attempting to turn on the tablet from power-off. I press the power button, the LED comes on, flashes extremely rapidly for a few seconds, and then goes dark, and the tablet stays off.

Sean-StarLabs, 2024-09-06:
@dchang0 Sounds right - that's it going from G3 to S5 (mechanically off to electronically off)

dchang0, 2024-09-06:
> @dchang0 Sounds right - that's it going from G3 to S5 (mechanically off to electronically off)

Hmm. Well, the intention is to turn the tablet on (from fully off to fully on), so it is puzzling.
I end up having to try a few times, and then it finally gets the point and turns on.

So, are you saying that the super-fast flashing of the power LED always means that the tablet is turning off electronically? If so, then that absolutely fits the OP's symptoms, but it is weird for my situation.

Sean-StarLabs, 2024-09-13:
> Hmm. Well, the intention is to turn the tablet on (from fully off to fully on), so it is puzzling.
I end up having to try a few times, and then it finally gets the point and turns on.

It's there to prevent accidentally turning it on - simply put, it's a double-press if there isn't a charger connected.

> So, are you saying that the super-fast flashing of the power LED always means that the tablet is turning off electronically? If so, then that absolutely fits the OP's symptoms, but it is weird for my situation.

No, flashing power LED is changing power states.

Sean-StarLabs, 2024-10-17:
Fixed with d6de2690dd547a014309086a9cc435e2b58f091e

closed 2024-10-17 by Sean-StarLabs