StarLabsLtd/firmware #212: [StarLite Mk V - Magnetic Keyboard] Insert key is missing

ggdaynight, 2024-09-14:
**Description of the issue:**

The Insert key is missing from the StarLabs StarLite Mk V Magnetic keyboard. I have not been able to invoke it using any key combination such as Fn + Del.

I am unsure if this should be treated as a separate issue or if it is related to [issue #194](https://github.com/StarLabsLtd/firmware/issues/194), which also discusses keyboard key mapping.

**System Information:**
- Model: StarLite Mk V
- OS: Manjaro with Gnome (pre-installed)
- Kernel version: 6.6.47-1-MANJARO
- Firmware version: 24.07 (updated)

**Additional information:**
I’ve reviewed the previous issues, read through the [Guides](https://support.starlabs.systems/kb/guides) provided by StarLabs, including the [Function keys guide](https://support.starlabs.systems/kb/guides/how-to-use-the-function-keys), but found no solution for invoking the Insert key. Additionally, I attempted to troubleshoot using the `libinput debug-events --show-keycodes` command with no success.

**Question:** Is it possible to add the Insert key functionality via a firmware modification or update?

Sean-StarLabs, 2024-09-16:
We don't have an insert key, but you can add/adjust it in Linux with xbindkeys/xmodmap

closed 2024-09-23 by Sean-StarLabs

ggdaynight, 2024-09-23:
Apologies for the delayed response - I only had time over the weekend to test this out.

While xbindkeys and xmodmap are viable solutions for X11, I’d like to ensure that I can use the Insert functionality in other contexts, such as TTY and Wayland (which I plan to switch to in the future).

I also examined the different Fn key behaviors in general. On some notebook keyboards, the Fn key may register with the OS, enabling custom key combinations like Fn+Del. However, on the StarLite Mk V Magnetic keyboard, the Fn key doesn’t seem to register with the OS, making it impossible to configure Fn+Del as a key combination for activating the Insert key. Given the already limited number of keys (78 keys in total), it’s also difficult to sacrifice another key to behave as the Insert key without affecting other functions.

With this in mind, I’m still interested in whether it would be possible to implement the Insert key as a Fn+Del combination directly through a firmware update. This would make the StarLite Mk V’s keyboard more consistent across platforms, including TTY, Wayland, and various operating systems, without relying on other tools.

Would you consider implementing this in the near future?

Thanks again for your help!

Sean-StarLabs, 2024-09-23:
> With this in mind, I’m still interested in whether it would be possible to implement the Insert key as a Fn+Del combination directly through a firmware update.

Yes

> Would you consider implementing this in the near future?

Hidden/unmarked key combos have not gone down well in the past, so we wouldn't add them unless a few people wanted them.

ggdaynight, 2024-09-23:
Thanks. I will ask around if there is any interest or if someone has a good solution for this.