StarLabsLtd/firmware #214: [StarLite Mk V] Add option to disable extremely bright led's

DevenBL, 2024-09-25:
Please add an option in the bios or somewhere else to disable the external led's

Sean-StarLabs, 2024-09-25:
Which ones? Why?

DevenBL, 2024-09-25:
The top LED near the power button is extremely bright at night. To have it either solid or pulsating makes sleep difficult if the tablet is on or suspended in the same room.
Potentially the charging port led as well although that does not currently seem to be as much of an issue.

Anachron, 2024-09-27:
I agree. I wish all devices that can be used in the sleeping room can have their LEDs turned down/off.

Sean-StarLabs, 2024-10-07:
Do you care if it does the "breathing"? It'll be possible to do, but I don't know how off the top of my head. Simple on/off for S3 with limited brightness is easy.

mixmastamyk, 2024-10-12:
Yes, too bright and too white.  Dim and/or orange/red would be better.

Unfortunately I also use it to know how long I've held the power button (blinks at 2 secs I think) to get it to come out of suspend.  So please allow the blink to happen even if you have a setting to turn it off.

gzagatti, 2024-10-27:
It would be great to be able to adjust this light. It is indeed too bright.

If no customization is possible, I would at least like to be able to switch it off completely as it is very distracting.

DevenBL, 2024-10-29:
An option to make the light not on all the time/not at max brightness seems to be the common point here. Some people however seem to want to retain the information conveyed by the light. 
Although i do not particularly care about the specifics of the implementation: 
Perhaps the option should be to have the power button light normally off and only communicate information through light when pressed? 

Fundamentally the problem is that the lights are on all or most of the time. Whatever solution you pick so long as there is an option to have the lights are on less in some way shape or form it would be an improvement.
What even is the best practice for this? On the steamdeck valve just allows you to turn the led off and you do things blindly. However since the led is also used when updating the firmware you would likely not want to disable the triple flash functionality when upgrading.

Sean-StarLabs, 2024-11-04:
Fixed with 6287b93e88154b7013df782a2e57405f403af339

closed 2024-11-04 by Sean-StarLabs

thebatdan, 2025-01-31:
I've updated to firmware 25.01 from LVFS, the bright LEDs seem to be back and the option to change the brightness seems to have disappeared from the Bios. Has there been a regression on this?

Sean-StarLabs, 2025-01-31:
Missed in upstreaming - will have it back in the next release:
https://review.coreboot.org/c/coreboot/+/86240