StarLabsLtd/firmware #216: [StarBook mkVI] ACPI: battery: [Firmware Bug]: (dis)charge rate invalid.

sim31, 2024-09-28:
I see this in logs on every boot:

```
kernel: ACPI: battery: [Firmware Bug]: (dis)charge rate invalid.
```

Wondering what this means and is it safe to ignore?

Sean-StarLabs, 2024-09-30:
What kernel, fw versions, charge level, and is the charger connected?

sim31, 2024-09-30:
Kernel: 6.10.10
Coreboot 24.03
EC: 24.3
Charge level is around 80%. The max charge level is set to 80%.

Charger is typically connected (USB-c) but today I tried booting without charger connected and I see the same warning.

Sean-StarLabs, 2024-09-30:
Ah yeah, fixed that a little while ago - fix in the next release, but it's not major

sim31, 2024-10-02:
Thanks. Looking forward to the next release.

Sean-StarLabs, 2024-10-11:
Fixed with adf1039e83eeb8f32c773488cf21d474c53ff8af

closed 2024-10-11 by Sean-StarLabs