StarLabsLtd/firmware #218: Starbook MK VI: charger stops working

sim31, 2024-10-11:
I bought this starbook with charger and usb-c cable from starlabs. Initially it was working fine. Later I started noticing charging status toggling on and off. Sometimes the charging would stop completely until I re-plugged. I tried it with a different charger. It seemed to resolve the issue. So I just bought a new charger and started using starbook with it (same cable).

Fast-forward about a year or so I noticed charging stopping out of nowhere again. Again re-plugging fixed it. It didn't happen again for a month or two. Now over the last week or so, it happened again - twice (maybe a day in between). Perhaps that's premature, but now I'm starting to think it's starting to go down the same road as with the old charger...

I found a bunch of comments from different people complaining about similar issues (you could find more if you included other models):

- https://github.com/starlabsltd/firmware/issues/197#issuecomment-2238482919
- https://github.com/StarLabsLtd/firmware/issues/95 
- https://github.com/StarLabsLtd/firmware/issues/160
- https://github.com/StarLabsLtd/firmware/issues/152
- https://github.com/StarLabsLtd/firmware/issues/153

The last few links made me remember that at least one of those times when charging stopped the laptop was under a pretty high load.

Sean-StarLabs, 2024-10-11:
Which StarBook (Intel 12th or 13th Gen or AMD)? Which firmware versions?

sim31, 2024-10-11:
12th gen Intel
Coreboot 24.03
EC: 24.3

Sean-StarLabs, 2024-10-11:
What's the make/model of the charger?

sim31, 2024-10-11:
https://eu.ugreen.com/products/ugreen-65w-usb-c-charger-with-3-ports?variant=40296225308755

Sean-StarLabs, 2024-10-11:
Fixed with adf1039e83eeb8f32c773488cf21d474c53ff8af

closed 2024-10-11 by Sean-StarLabs

sim31, 2024-11-19:
It started happening more often.

@Sean-StarLabs , what was the nature of the issue that was fixed? Currently, it seems to me that the charger gradually deteriorates. Could the issue have caused something like that?

Sean-StarLabs, 2024-11-19:
What version are you running? How did you install it? What settings are you using?

The issue was NTC calculations not happening fast enough; not sure what you mean by "degrades"?

sim31, 2024-11-19:
Version: 24.09, but as I was implying above it was happening with previous versions as well. I installed it from ubuntu 23.10 using gnome-firmware app. I use arch linux, but dual booting with ubuntu for installing firmware, because fwupd did not work properly from arch (might be working again).

Regarding settings, I have max charging level set to 80%. Everything else is default.

By "degrades" I mean that it is getting worse, and less usable. As I mentioned above, this is second charger this is happening to. With the first one, it was fine at first, then happened a couple of times, then it started happening more and more often. This charger worked seemed to have fixed it, but then it happened a couple of times, and now seems to start happening more often again.

Sean-StarLabs, 2024-11-19:
There's no point looking into old releases, please update

> By "degrades" I mean that it is getting worse, and less usable. As I mentioned above, this is second charger this is happening to. With the first one, it was fine at first, then happened a couple of [...]

Gotcha - to answer then, no, it's not that.

sim31, 2024-11-21:
I updated to 24.10. It already happened once again.

Now, experienced another issue that might not be related (let me know if I should create another issue): reported battery level is going up even though it's disconnected from power. If I look at power management settings in gnome, the remaining estimated time there is jumping between 2 hours to 5. Also, yesterday at one point the power led was purple even though battery level reported by the OS was 60% (max charging level is set to 80%). It was back to 80% after a reboot.

Sean-StarLabs, 2024-11-21:
> I updated to 24.10. It already happened once again.

With the EC on 24.10 too? Can you provide more info?

>  experienced another issue

That's not an issue; runtime is a variable, it'll decrease or increase depending on usage.

Sean-StarLabs, 2024-11-21:
> Could it be because of a restart instead of fresh boot? 

It's nothing to do with the boot cycle, but more giving the chance for the EC to update. When coreboot and  the EC aren't in sync, weird things happen as offsets are wrong, old values are held etc. Worst case is the firmware gets bricked, usually  it's just minor annoyances.

So disregard the behaviours that you saw, as we have to assume it's because a step of the update was skipped, and see how it goes from now.