StarLabsLtd/firmware #219: Alert on UEFI dbx lite mkIII

soc7, 2024-10-14:
I am seeing an alert after updating the UEFI dbx on the lite mkIII as following:

UEFI Secure Boot Forbidden Signature Database

This updates the dbx to the latest release from Microsoft which adds insecure versions of grub and shim to the list of forbidden signatures due to multiple discovered security updates.

And it poits those CVEs as issues:
CVE-2020-7205
CVE-2020-15707
CVE-2020-15706
CVE-2020-15705
CVE-2020-14311
CVE-2020-14310
CVE-2020-14309
174059
CVE-2020-14308
CVE-2020-10713

Should I be worried?

Sean-StarLabs, 2024-10-14:
> Should I be worried?

What are you asking?

soc7, 2024-10-14:
How serious are those vulnerabilities?

Sean-StarLabs, 2024-10-14:
Real-world, likely not very.  Severity scores are more of a "what could happen".

soc7, 2024-10-14:
Like only physical attack? No remote attack?

Sean-StarLabs, 2024-10-14:
I don't know - most require some level of access to exist before they're an issue

samueldr, 2024-10-14:
> Should I be worried? [about UEFI dbx updates]

Those updates only affect use of Secure Boot. They “revoke” [actually deny-list] public keys from being allowed for use with Secure Boot enabled.

 - They do not change the UEFI for your system
 - They do not fix bugs in the UEFI for your system
 - They prevent vulnerable (buggy) ***Secure Boot signed binaries*** from running when Secure Boot is enabled.

* * *

Since your initial message doesn't clearly state whether ***you*** are using Secure Boot, it depends:

First of all, like Sean said, they all pretty much require some form of access to the system (physical access, or other vulnerabilities to exploit) so boot entries can be modified, or added.

**If you are using Secure Boot, *and* rely on Secure Boot for its properties**: maybe, you could be worried. Not updating the revocation list means known insecure signed binaries can still run on your system at the bootloader stage.

**If you are not using Secure Boot (e.g. by disabling it)**: no, because whether signed or not, any arbitrary binaries can already be booted on your system.

closed 2024-10-17 by Sean-StarLabs