StarLabsLtd/firmware #220: [StarLite Mk V] update 24.08 firmware, dock status and volume buttons not working

mixmastamyk, 2024-10-15:
Hi, I updated to 24.08 but have a few issues.  
(Latest Fedora, 6.10.x series.)

### Volume buttons no longer working under Linux.

These buttons used to work in Linux.  They *still* work in the firmware setup.  And the power/nap button still works everywhere.  But no volume!  Need to set from control panel now.  :-(

Where to look for this kind of stuff?  I went down a rabbit hole with i2c but that seems to be the touch screen.  Are these intel virtual buttons?

```
⏵ sudo libinput debug-events --show-keycodes
-event1   DEVICE_ADDED            Power Button                      seat0 default group1  cap:k
-event7   DEVICE_ADDED            Video Bus                         seat0 default group2  cap:k
-event0   DEVICE_ADDED            Lid Switch                        seat0 default group3  cap:S
-event6   DEVICE_ADDED            GXTP7386:00 27C6:0111 Keyboard    seat0 default group4  cap:k
-event3   DEVICE_ADDED            GXTP7386:00 27C6:0111             seat0 default group4  cap:t  size 268x179mm ntouches 10 calib
-event4   DEVICE_ADDED            GXTP7386:00 27C6:0111 Stylus      seat0 default group4  cap:T  size 268x179mm left calib
-event9   DEVICE_ADDED            Intel Virtual Buttons             seat0 default group5  cap:k
-event10  DEVICE_ADDED            Intel Virtual Switches            seat0 default group6  cap:S
 event10  SWITCH_TOGGLE           +0.000s	switch tablet-mode state 1
-event8   DEVICE_ADDED            Intel HID events                  seat0 default group7  cap:k
-event2   DEVICE_ADDED            AT Translated Set 2 keyboard      seat0 default group8  cap:k

 event1   KEYBOARD_KEY            +4.515s	KEY_POWER (116) pressed
 event1   KEYBOARD_KEY            +4.515s	KEY_POWER (116) released
```

Power button and touch screen work, volume buttons don't respond.  I attached a keyboard and its volume keys work.

### Dock status no longer available under Phosh.

The top status menu used to have a toggle for dock/undocked status.  It no longer does, and so I cannot change profiles any longer.  This status controls things like onscreen keyboard, showing mobile friendly apps, etc.

This used to enable it, when chassis field was set to convertible.  I tinkered with this command before:

```
⏵ hostnamectl
     Static hostname: starlite
           Icon name: computer-convertible
             Chassis: convertible
```

-- 

### Sleep fix worked.

I also created a script to fix the sleep issue, (the echo/tee command  #183), and enabled it.  Unfortunately at the same time, so not 100% sure it is unrelated.  However I have now disabled it, shutdown, and restarted and volume keys and/or dock status still not working.

I went back to an older kernel as well, didn't help.

mixmastamyk, 2024-10-15:
This seems to be the device that's ignored:

```
⏵ sudo udevadm info -p /sys/class/input/event2
[sudo] password for mgmiller: 
P: /devices/platform/i8042/serio0/input/input2/event2
M: event2
R: 2
U: input
D: c 13:66
N: input/event2
L: 0
S: input/by-path/platform-i8042-serio-0-event-kbd
E: DEVPATH=/devices/platform/i8042/serio0/input/input2/event2
E: SUBSYSTEM=input
E: DEVNAME=/dev/input/event2
E: MAJOR=13
E: MINOR=66
E: USEC_INITIALIZED=3962383
E: ID_INPUT=1
E: ID_INPUT_KEY=1
E: ID_INPUT_KEYBOARD=1
E: ID_BUS=i8042
E: ID_SERIAL=noserial
E: ID_PATH=platform-i8042-serio-0
E: ID_PATH_TAG=platform-i8042-serio-0
E: LIBINPUT_DEVICE_GROUP=11/1/1:isa0060/serio0
E: DEVLINKS=/dev/input/by-path/platform-i8042-serio-0-event-kbd
E: TAGS=:power-switch:
E: CURRENT_TAGS=:power-switch:
```

mixmastamyk, 2024-10-16:
Not sure what's going on.  Volume keys work in:

- Firmware setup
- Linux console (shows escape sequences at prompt)
-  sudo evtest

Unfortunately Wayland/libinput refuse to acknowledge them for some reason.  Thought it might be a recent OS update, but I rolled back all the way to last month and it still doesn't work.

May have to go back to old firmware, but I've never done that before.  Will look into it tomorrow.

Sean-StarLabs, 2024-10-16:
They're PS2 - simply sending scan codes, re-checked literally yesterday on all of the distros we provide an their happy so I'd assume something software-wise is discarding them

mixmastamyk, 2024-10-16:
There may be other issues, but specifically in regards to my problem—I don't have the starlite keyboard, nor is it related to suspend.  Volume buttons are not acknowledged in the Wayland/GUI.

Sean-StarLabs, 2024-10-16:
No thank you very much - I have no idea what swaywm is :)

Forgetting S3, if you boot without a keyboard - is it the same behaviour?

mixmastamyk, 2024-10-16:
Yes, I am typically using as a tablet, without keyboard.

I just tried it with X11/Cinnamon and they work there, so must be phosh or wayland... narrowing it down.

mixmastamyk, 2024-10-16:
Welp, downgrading firmware to 24.07 fixed the volume button problem with Wayland/Phosh.  Why?  No idea when it worked everywhere else.  Maybe some kind of naming mismatch.

Sean-StarLabs, 2024-10-16:
Thanks - I know what is... how to fix is another story

mixmastamyk, 2024-10-16:
The plot thickens... I just downgraded the firmware back to 24.06 and got my manual dock toggle button back.  Looks like I'm sticking to 24.06 for a while.

Dunno what it is looking at to decide to surface that button, and docs aren't that great.  May need to dig thru the source to find out.  Unfortunately I've spent a few full days on this and better get back to work.  :-D

Sean-StarLabs, 2024-10-16:
>  got my manual dock toggle button back.

It looks at the dock status - not a bug, how it should be. DE add-ons exist to make it ignore the dock status.

mixmastamyk, 2024-10-16:
The toggle button is useful because the system otherwise doesn't differentiate when I plug in my hub with peripherals.  Leading to the onscreen keyboard coming up when a physical one is available, etc.

Sean-StarLabs, 2024-10-17:
Fixed with d6de2690dd547a014309086a9cc435e2b58f091e

closed 2024-10-17 by Sean-StarLabs

mixmastamyk, 2024-10-17:
No such add-ons exist for phosh, that I know of.

If the dock status mechanism were documented somewhere it would be helpful.  So far I have no idea how it works and can't find anything on it.

Sean-StarLabs, 2024-10-18:
Patch - https://github.com/StarLabsLtd/coreboot/commit/bc3a02f309f5e71e888150b039a43f4e84923458

Might help - https://github.com/shyzus/gnome-shell-extension-screen-autorotate

mixmastamyk, 2024-10-21:
Hmm, looks like it is provided by GPIO pins at the bottom of the tablet, i.e. connected to your keyboard case only.  

Hopefully software can handle multiple dock instances, say over USB-C.

Sean-StarLabs, 2024-10-21:
AFAIK, it doesn't because there's only ever going to be one dock - hubs and plugged in things are different. The GNOME one just adds another instance of the button.