StarLabsLtd/firmware #221: [StarLite MK V] Bad performance on BIOS 24.09

bergmannf, 2024-10-21:
After upgrading to the newest bios using LVFS (24.09) the performance of the tablet has gotten way worse.
It seems everything is using 100% CPU immediately and everything feels very slow compared to 24.08.

I downgraded after a few minutes, because even just browsing the web felt sluggish and even using a terminal was noticeably worse than with 24.08.

Sean-StarLabs, 2024-10-21:
With what settings?

bergmannf, 2024-10-21:
If you are referring to the BIOS settings - I only changed the charging behavior to 0.5C when upgrading to 24.08, but then never touched them again - is there anything that needs to be adjusted?

Sean-StarLabs, 2024-10-21:
Nothing "has" to be adjusted, but it's like saying your car isn't going fast enough and you haven't taken it out of first gear,

bergmannf, 2024-10-21:
I checked the 24.09 options - the performance was set to 'Balanced' same as before and RAM speed to 5500 MT/s (I guess that toggle is new as I don't remember seeing it before).
I did see the `Reworked performance profiles for race to idle` - but I don't see anything related to this in the setting, so I guess those are just changes to the existing performance profile? (Which wasn't set to be used before, so I didn't change to it either)

s3ph-scott, 2024-10-21:
I've been having similar issues. Due to needing the tablet for work I had to revert back to 24.08 so my analysis of the issue is incomplete.

I am using Ubuntu 24.04.1 with Kernel 6.8. My device is encrypted with LUKS and I have the 2TB SSD.

Just as an example if I load a YouTube video in Firefox 131 I get a significant amount of dropped frames to the point where the video stops itself then starts again (resets the dropped frames to 0). I also see significant interface lag in Ubuntu in general as well as perceivable longer times getting to the login screen after presenting the password to  unlock the volume.

To try and fix it I reset the Coreboot settings to default but this did not help.

I've also switched the Ubuntu settings from power saver to high performance but this did not help.

If I run the _top_ command I note some processes will go straight to 100% CPU for a long time. There is plenty of memory free so I don't think there's a problem there.

As I've wrote this post there's two things I need to check next time I have time:

- What speed the CPU itself is running at i.e. is it down clocking aggressively.
- If detaching the cover has any bearing.

I did also notice two things about 24.09 in the Coreboot menu I don't think are related and would need to confirm I read the values right:

- There are now three options for memory speed but they are all somehow higher than the specification page quotes at 4800 mt/s.
- Memory reports 0mb.

DevenBL, 2024-10-22:
Same issue here. Although Changing settings from Balanced to Performance is a workaround for this issue on 24.09 with it set to Performance mode the behavior is similar to 24.08 Balanced mode

closed 2024-10-22 by Sean-StarLabs

bergmannf, 2024-10-22:
I ran a geekbench benchmark with 24.08 and 24.09 using the Balanced and Performance profiles each time - it shows that 24.09 with Balanced is extremely low: 50% lower on single core scores and only ~30% of the 24.08 results on multicore.
24.08 Balanced: https://browser.geekbench.com/v6/cpu/8434229
24.08 Performance:   https://browser.geekbench.com/v6/cpu/8434961
24.09 Balanced: https://browser.geekbench.com/v6/cpu/8435243
24.09 Performance: https://browser.geekbench.com/v6/cpu/8435404

DevenBL, 2024-10-24:
> I ran a geekbench benchmark with 24.08 and 24.09 using the Balanced and Performance profiles each time - it shows that 24.09 with Balanced is extremely low: 50% lower on single core scores and only ~30% of the 24.08 results on multicore. 24.08 Balanced: https://browser.geekbench.com/v6/cpu/8434229 24.08 Performance: https://browser.geekbench.com/v6/cpu/8434961 24.09 Balanced: https://browser.geekbench.com/v6/cpu/8435243 24.09 Performance: https://browser.geekbench.com/v6/cpu/8435404
> 
> Not sure if that's possible, but given the big difference maybe the 24.09 default should be set via the update to Performance then? I wouldn't expect having to change something in my BIOS when all I wanted was the bluetooth fixes.

just for additional information because i looked into it.
https://github.com/StarLabsLtd/coreboot/commit/b115ca0eb24d2eb1775493e18a645d12f2fdd92c
looking at the commits for coreboot
yeah seems like that ~30% reduction in performance was very much intended.

Performance as the default would probably give a better first impression of the tablet.
and while i could submit a pull request to change src/mainboard/starlabs/starlite_adl/cfr.c line 149 from a 1 to a 2 to make it the default. I have a feeling submitting the world's smallest pull request to change it would be bad form.

Sean-StarLabs, 2024-10-24:
> yeah seems like that ~30% reduction in performance was very much intended.

Changes are intentional, but ~30% isn't right, as you're only looking at one patch, not factoring tau or tcc etc.

> I have a feeling submitting the world's smallest pull request to change it would be bad form.

coreboot doesn't use pull requests. Defaults are picked based on number of requests - nothing else.

pdfrod, 2024-10-27:
Same issue here. Just downgraded to 24.08 because 24.09 left my machine painfully slow. I tried "performance" power mode, but that didn't help much; performance was still really poor.

For example, something as simple as opening the terminal can take seconds with 24.09, while with 24.08 it's instantaneous.

DevenBL, 2024-10-28:
Just to be clear you were changing the performance mode in the bios/uefi correct?

MRDGH2821, 2024-10-28:
Here are some of my observations about daily light weight usage.

This is for 24.09 with liquorix kernel:
| BIOS Power profile | Memory speed | OS Power profile | Notes                                                                                                                                                                                       |
| ------------------ | ------------ | ---------------- | ------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------- |
| Performance        | 6400 MT/s    | Performance      | Performance was not good. KDE's UI was lagging. Opening up some apps triggers 100% CPU utilisation. Tablet wasn't responsive that much. But I may need to recheck as I don't remember much. |
| Performance        | 6400 MT/s    | Power save       | Performance was excellent. Everything was responsive even if there was slight lag.                                                                                                          |
| Balanced           | 5500 MT/s    | Performance      | Same as Performance-Performance                                                                                                                                                             |
| Balanced           | 5500 MT/s    | Power save       | Same as Performance-Power save                                                                                                                                                              |

Sean-StarLabs, 2024-10-28:
"Good" is a hard matric to quantify - if you're comparing against 24.08, then a fair comparison would be with the memory at 7500 MT/s