StarLabsLtd/firmware #222: [StarLite Mk V] BIOS version `24.09` breaks touch screen, active pen, and plymouth; `24.08` "just works"

sodiboo, 2024-10-21:
I use NixOS and my exact reproducible system configuration is available at [`github:sodiboo/system`](https://github.com/sodiboo/system), but that seems irrelevant because i have observed the following:

Previously, my boot times were taking a really long time. So i disabled plymouth to figure out what it is, found it's the TPM failing to intialize, and i saw this was fixed in the BIOS update. Naturally, [i compiled `fwupd` with the `flashrom` plugin, so that i could update the BIOS](https://github.com/sodiboo/system/commit/7c602501f7308fac5851f4b764756d6bce43b26e). Prior to this i was on BIOS and EC version `24.06` and `24.6` (EC doesn't have a leading zero?).

I then upgraded to BIOS `24.09`, after which my tpm issue was fixed. I enabled plymouth again, but now it no longer masks the boot output, and instead it seems to only freeze parts of it? Previously, this would show the StarLabs logo, but now it shows systemd output.

I also noticed this today: my touchscreen no longer works! and neither does my "MPP Active Pen". This is unacceptable, as it's like the primary reason i bought a StarLite.

But i figured since i'd just updated my BIOS, that might have had something to do with it. So, i downgraded my BIOS again with `fwupdmgr`, which went very smoothly. I've also said "yes, send details" every time it asks me to upload analytics; i'm not sure how much can be correlated in the vendor backend side of that mechanism, but if device identifiers are included, the upgrade `24.06->24.09` should have happened a few days ago and then today `24.09->24.08`. I stated in the commit i linked that my EC is stuck on 24.6 and i didn't know how to update it. Turns out, this is not true, it has been updated since i wrote that, but since i thought it was unupdatable with my setup i haven't kept track of its version throughout.

I noticed on `24.08` that my touchscreen works again. So does the pen. Plymouth also works, which i didn't expect it to suddenly start working. I figured, maybe this is a fluke? I upgraded to `24.09`, and i observed the same issues: my touchscreen didn't register touches or pens, and Plymouth is unable to paint over systemd in the boot sequence.

I then downgraded again to `24.08`, and those things began to work again.

Despite [the release notes](https://github.com/StarLabsLtd/firmware/tree/main/StarLite/MkV) mentioning that the TPM issue was fixed in `24.09`, i haven't had the issue on `24.08` either, only ever on `24.06`. I suspect this is because when i checked the `/sys/class/dmi/id/ec_firmware_release` now, it shows `24.9`. I have no idea what values it had throughout my upgrade/downgrade cycle, except at the start when it said `24.6` (despite being on bios `24.09`).

---

From the above described rotation of BIOS versions where i've tested both of these versions twice (going `24.06->24.09->24.08->24.09->24.08`), i can with reasonable confidence state that:

On `24.08`:

- Plymouth can draw over the boot screen
- Touch inputs register in my compositor
- Pen inputs register in my compositor.

On `24.09`:

- Plymouth can't draw over the boot screen
- Touch inputs don't register in my compositor
- Pen inputs don't register in my compositor.

The TPM issues causing a boot timeout are apparently fixed on *both* of these BIOS versions, despite being part of `24.09`'s changelogs.

---

Plymouth doesn't fail, it just doesn't render anything. When plymouth isn't working, i do see the systemd output that it's going to "Terminate Plymouth". which is ironic. because plymouth should stop me from seeing that line of output.

Sean-StarLabs, 2024-10-21:
> i didn't know how to update it

It'll pick the EC up from coreboot, but to apply it, it needs to reach G3 (mechanically off) - just turn it off, disconnect the charger wait 12 seconds. Next time you connect the charger, it'll use whatever EC is in coreboot.

> the TPM issue was fixed in 24.09

No, that's something else - I don't think anyone else has mentioned this.

> Plymouth is unable to paint over systemd in the boot sequence.

Paint?

> my touchscreen no longer works! 

Please post a dmesg

DevenBL, 2024-10-21:
Confirming the issue as i am experiencing the same. 
[dmesg.log](https://github.com/user-attachments/files/17467306/dmesg.log)
here is my personal dmesg log

sodiboo, 2024-10-21:
> just turn it off, disconnect the charger wait 12 seconds. Next time you connect the charger, it'll use whatever EC is in coreboot.

Yeah, that seems accurate. did this, and now it says EC is at `24.8`.

> No, that's something else - I don't think anyone else has mentioned this.

Interesting. It works on 24.08 and 24.09, so I'm not sure it's worth looking a lot into it, but if you do want to investigate the issue, the message I saw in the boot sequence is "a start job is running for /dev/tpmrm0 (x sec / 1 min 30 sec)", which would eventually hit the time limit and fail to start, at which point the boot sequence can continue and I'm finally dropped into a greeter. I think that sounds like what was mentioned in the release notes. An exactly reproducible build of an OS that encounters this fault on `24.06` firmware should be available in my system configuration, in the commit prior to adding flashrom support to `fwupd`

> Paint?

as in, Plymouth takes over the display during the boot sequence. It "paints" some simple graphic based on the theme you define, as well as a loading spinner. on `24.09` the systemd output looks as if Plymouth is working fine; but if Plymouth was working fine, I wouldn't be able to see the output in question, as it would paint over systemd until the system reaches the greeter. Plymouth does work fine on BIOS `24.08`, when the OS is otherwise entirely unchanged.

> Please post a dmesg

How do I do this? What command do I run to get the output you want?

DevenBL, 2024-10-22:
> > Please post a dmesg
> 
> How do I do this? What command do I run to get the output you want?

I assume he wants the output of `sudo dmesg`

Sean-StarLabs, 2024-10-22:
[coreboot.zip](https://github.com/user-attachments/files/17475469/coreboot.zip)

Thanks - can you try this? Download, extract, `sudo flashrom -p internal -w coreboot.rom -i bios --ifd -n -N` then *shutdown*.

DevenBL, 2024-10-22:
tried it all issues persist on my end. 
[dmesg2.log](https://github.com/user-attachments/files/17481817/dmesg2.log)
here is updated dmesg log.  pretty much the same behavior as sodiboo

sodiboo, 2024-10-26:
The provided build fixes plymouth, and it can properly obscure the boot logs. It seems to make everything else worse, though. The boot time (measured by `dmesg`) on `24.08` is about 10 seconds, on the released `24.09` it's about 40 seconds, and on this build it's closer to 100 seconds. Pen and touch still don't work on this build, nor do they on released `24.09`, but still work fine on `24.08`.

This time, now knowing how to make the EC update, i made sure to unplug AC when shutdown after an update, and these dmesg logs are with matching BIOS *and* EC firmware. I got logs from two different boots again, just in case anything was a fluke (but it doesn't seem to have been a fluke)

https://gist.github.com/sodiboo/aba45b8414aaec2544fe208160af546d

DevenBL, 2024-11-03:
24.10 seems to fix these issues for me.

Sean-StarLabs, 2024-11-04:
Fixed with 6287b93e88154b7013df782a2e57405f403af339

Sean-StarLabs, 2024-11-04:
> 24.10 seems to fix these issues for me.

I only built the release version an hour ago.

DevenBL, 2024-11-04:
> > 24.10 seems to fix these issues for me.
> 
> I only built the release version an hour ago.

so should i reinstall? since i used a pre release 24.10 i guess?

Sean-StarLabs, 2024-11-05:
Installing stable releases is safer, but it's up to you