StarLabsLtd/firmware #223: Kernel doesn’t show correct bios version

tensor5, 2024-10-26:
Upgraded to 24.09:

```console
$ uname -r
6.11.5

$ cat /sys/class/dmi/id/bios_release
24.8

$ cat /sys/class/dmi/id/ec_firmware_release
24.9
```

Sean-StarLabs, 2024-10-27:
How did you check 24.09 was installed?

MRDGH2821, 2024-10-28:
Strange even I'm seeing this even though I explicitly flashed 24.09 in my Starlite MK V

MRDGH2821, 2024-10-28:
I rebooted & cross checked in firmware settings, BIOS version is indeed 24.09. But one peculiar thing I noticed is it shows 0 MB RAM instead of 16 GB (equivalent in MB) RAM

Sean-StarLabs, 2024-10-28:
Ah, didn't see that at first - you're looking at the release, not version.

`cat /sys/class/dmi/id/bios_version`

tensor5, 2024-10-28:
> I rebooted & cross checked in firmware settings, BIOS version is indeed 24.09. But one peculiar thing I noticed is it shows 0 MB RAM instead of 16 GB (equivalent in MB) RAM

Same here.