StarLabsLtd/firmware #228: [Error in build] unknown escape sequence in src/soc/intel/common/block/cnvi.c

Hacksawfred3232, 2024-11-11:
An error has been introduced into src/soc/intel/common/block/cnvi/cnvi.c
```
src/soc/intel/common/block/cnvi/cnvi.c: In function 'cnvw_fill_ssdt':
src/soc/intel/common/block/cnvi/cnvi.c:214:97: error: unknown escape sequence: '\_' [-Werror]
  214 |                                                         acpigen_emit_namestring("\_SB.PCI0.BTRK");
      |                                                                                                 ^
src/soc/intel/common/block/cnvi/cnvi.c:254:113: error: unknown escape sequence: '\_' [-Werror]
  254 |                                                                         acpigen_emit_namestring("\_SB.PCI0.BTRK");
      |                                                                                                                 ^
cc1: all warnings being treated as errors
make: *** [Makefile:430: build/ramstage/soc/intel/common/block/cnvi/cnvi.o] Error 1
```
Using a double slash instead of a single slash fixes the issue for now, but may want to look into it to see if this was an error or if was intended and GCC is throwing a wobbly.

Sean-StarLabs, 2024-11-11:
Should be `\\`, not sure where you're building from but stuff like that is picked up by Jenkins

Hacksawfred3232, 2024-11-11:
```
> git status
On branch 24.09
Your branch is up to date with 'origin/24.09'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
	modified:   src/soc/intel/common/block/cnvi/cnvi.c

Untracked files:
  (use "git add <file>..." to include in what will be committed)
	.custom/

no changes added to commit (use "git add" and/or "git commit -a")
```
.custom contains Logo.bmp and flashrom.sh (an automation script). Otherwise, I have a 1:1 copy of the github repo, tracked on branch 24.09. The problematic lines shown in the error are in branch shown:
https://github.com/StarLabsLtd/coreboot/blob/24.09/src/soc/intel/common/block/cnvi/cnvi.c#L214
https://github.com/StarLabsLtd/coreboot/blob/24.09/src/soc/intel/common/block/cnvi/cnvi.c#L254

Hacksawfred3232, 2024-11-11:
Ah, hang on, 24.09 - the default branch - has the error, but 24.10 does not. I'm guessing 24.10 is currently WIP?

Sean-StarLabs, 2024-11-11:
If you're after stable, reviewed code - point at gerrit