StarLabsLtd/firmware #232: starlabs-touchpad is not available

jozic, 2024-12-22:
according to https://support.starlabs.systems/kb/known-issues/unable-to-right-click-or-drag-windows one should be able to install starlabs-touchpad, but it's not available.

```
➜  ~ sudo add-apt-repository ppa:starlabs/main
You are about to add the following PPA:
 ███████╗████████╗ █████╗ ██████╗ ██╗      █████╗ ██████╗ ███████╗
██╔════╝╚══██╔══╝██╔══██╗██╔══██╗██║     ██╔══██╗██╔══██╗██╔════╝
███████╗   ██║   ███████║██████╔╝██║     ███████║██████╔╝███████╗
╚════██║   ██║   ██╔══██║██╔══██╗██║     ██╔══██║██╔══██╗╚════██║
███████║   ██║   ██║  ██║██║  ██║███████╗██║  ██║██████╔╝███████║
╚══════╝   ╚═╝   ╚═╝  ╚═╝╚═╝  ╚═╝╚══════╝╚═╝  ╚═╝╚═════╝ ╚══════╝

This PPA contains packages for the laptops sold and supported
by Star Labs including:

- StarLite
- StarLabTop
- StarBook
- StarFighter

They are only designed for these laptops. Using them on any other computer is at your own risk.
 More info: https://launchpad.net/~starlabs/+archive/ubuntu/main
Press Enter to continue or Ctrl+C to cancel

gpg: keybox '/etc/apt/keyrings/17A20BAF70BEC3904545ACFF8F21C26C794386E3.keyring' created
gpg: key 8F21C26C794386E3: public key "Launchpad PPA for Star Labs" imported
gpg: Total number processed: 1
gpg:               imported: 1                                                                                                                                                           /7.8s
➜  ~ sudo apt update                          
Hit:1 http://dl.winehq.org/wine-builds/ubuntu jammy InRelease
Hit:2 http://archive.ubuntu.com/ubuntu jammy InRelease                                                                                                                                       
Ign:3 http://packages.linuxmint.com virginia InRelease                                                                                                                                       
Hit:4 http://archive.ubuntu.com/ubuntu jammy-updates InRelease                                                                                                                               
Hit:5 http://security.ubuntu.com/ubuntu jammy-security InRelease                                                                                                                             
Hit:6 http://archive.ubuntu.com/ubuntu jammy-backports InRelease                                                                                                                             
Hit:7 http://packages.linuxmint.com virginia Release                                                                                                                                         
Hit:8 https://updates.signal.org/desktop/apt xenial InRelease                                                                                                            
Hit:9 https://deb.opera.com/opera-stable stable InRelease                                                                                          
Hit:10 https://dl.google.com/linux/chrome/deb stable InRelease                                                               
Hit:11 https://packages.microsoft.com/repos/code stable InRelease                                      
Hit:13 https://ppa.launchpadcontent.net/appimagelauncher-team/stable/ubuntu jammy InRelease
Get:14 https://ppa.launchpadcontent.net/starlabs/main/ubuntu jammy InRelease [18.0 kB]
Hit:15 https://ppa.launchpadcontent.net/ufleisch/kid3/ubuntu jammy InRelease
Get:16 https://ppa.launchpadcontent.net/starlabs/main/ubuntu jammy/main amd64 Packages [2,656 B]
Get:17 https://ppa.launchpadcontent.net/starlabs/main/ubuntu jammy/main i386 Packages [728 B]
Get:18 https://ppa.launchpadcontent.net/starlabs/main/ubuntu jammy/main Translation-en [2,336 B]
Fetched 23.8 kB in 2s (14.9 kB/s) 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.                                                                                                                                                             /2.6s
➜  ~ sudo apt install starlabs-touchpad
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package starlabs-touchpad                                                                                                                                            /0.2s
➜  ~ 

```

Sean-StarLabs, 2024-12-22:
Shouldn't be needed anymore - why are you after it?

jozic, 2024-12-22:
I tried upgrading my Linux mint from 21.3 to 22 (which is newer Ubuntu based), touchpad had the issue. Then I found this page. Eventually I had to roll back using timeshift. Just wanted to let you know the page is outdated and needs to be fixed or removed. But given my issue with the touchpad, I would prefer it to be fixed :)
Thanks for checking.

Sean-StarLabs, 2024-12-23:
Gotcha - it does need to be removed. You'll have something on your install that needs to go, maybe an old copy of starlabs-touchpad, that's causing issues. If you can't find it, feel free to ping support.

closed 2024-12-23 by Sean-StarLabs