StarLabsLtd/firmware #243: Audiojack not working with coreboot, but works with AMI on L3U

soc7, 2025-01-10:
### **Describe the bug**
Provide a clear and concise description of the issue, including any error messages, unexpected behaviors, or system crashes.
The audiojack is not working when running any version of coreboot on the Labtop mkIII

### **Steps to Reproduce**
Detailed steps to reproduce the issue:
1. Install coreboot
2. Insert a audiojack headphones 
3. It will never work neither microfone or headphone audio

Be as specific as possible to help us replicate the issue.

I am not very sure about that, my observasion is that after running coreboot for this device since the start and upgrading the releases until 24.10 I never got the audiojack to work, yesterday I flashed back to AMI and it worked.

### **Expected Behavior**
Describe what you expected to happen instead of the observed behavior.

Audiojack to work, when plugging in and playing songs etc. Also the microphone

### **Screenshots**
If applicable, upload screenshots or screen recordings to help visualize the issue.

---

### **System Information**
Provide your system details:
- **OS: Debian bookworm
- **Kernel Version: 6.1.119-1 (2024-11-22) 
- **Product SKU:** L3U
- **BIOS Version: 24.10 on coreboot and V2.0.00dme on AMI
- **EC Release: 3.12

---

### **Logs**
Include any relevant logs that might help diagnose the issue:
- **Dmesg Output:** Attach or paste the relevant portion of `dmesg` logs.
- **Other Logs:** Specify and include logs from related subsystems if applicable.

while running AMI:
12.829109] input: HDA Intel PCH Mic as /devices/pci0000:00/0000:00:1f.3/sound/card0/input14
[   12.829161] input: HDA Intel PCH Front Headphone as /devices/pci0000:00/0000:00:1f.3/sound/card0/input15
[   12.829201] input: HDA Intel PCH HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/sound/card0/input16
[   12.829236] input: HDA Intel PCH HDMI/DP,pcm=7 as /devices/pci0000:00/0000:00:1f.3/sound/card0/input17
[   12.829275] input: HDA Intel PCH HDMI/DP,pcm=8 as /devices/pci0000:00/0000:00:1f.3/sound/card0/input18

I will reinstall coreboot and then edit here with its logs.

### **Additional Context**
Provide any other information or context that might help in debugging this issue. This could include:
- Previous steps taken to resolve the issue.
- Workarounds attempted, if any.
- Frequency of the issue (e.g., "Always," "Intermittent").

---

soc7, 2025-01-12:
I am trying to switch branch back to coreboot 24.10 but I am getting the following error:
`$ fwupdmgr switch-branch
Dispositivo selezionato: LabTop (bios)
╔══════════════════════════════════════════════════════════════════════════════╗
║ Passare dal ramo predefinito a coreboot?                                     ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ General Updates:                                                             ║
║                                                                              ║
║ • Further improve the Bluetooth driver for lower power consumption           ║
║ • Interpolate charging current relative to 25C                               ║
║ • [Byte Mk II] Add more ACPI for the fTPM                                    ║
║ • [StarLite Mk V] Made it possible to lower the brightness of the power      ║
║ LED                                                                          ║
║                                                                              ║
║ Security Updates:                                                            ║
║                                                                              ║
║ • Enroll a default set of keys for SecureBoot                                ║
║                                                                              ║
║ Bug Fixes/Enhancements:                                                      ║
║                                                                              ║
║ • [StarBook Mk VI] Fix an issue with DSD generation                          ║
║ • [StarLite Mk V] Increase the reset delay for the touchscreen               ║
║ • Account for late OCM loading                                               ║
╚══════════════════════════════════════════════════════════════════════════════╝
Le conseguenze del cambio di ramo del firmware sono state comprese? [y|N]: y
╔══════════════════════════════════════════════════════════════════════════════╗
║ Aggiornare LabTop (bios) da 2.0 a 24.10?                                     ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ General Updates:                                                             ║
║                                                                              ║
║ • Further improve the Bluetooth driver for lower power consumption           ║
║ • Interpolate charging current relative to 25C                               ║
║ • [Byte Mk II] Add more ACPI for the fTPM                                    ║
║ • [StarLite Mk V] Made it possible to lower the brightness of the power      ║
║ LED                                                                          ║
║                                                                              ║
║ Security Updates:                                                            ║
║                                                                              ║
║ • Enroll a default set of keys for SecureBoot                                ║
║                                                                              ║
║ Bug Fixes/Enhancements:                                                      ║
║                                                                              ║
║ • [StarBook Mk VI] Fix an issue with DSD generation                          ║
║ • [StarLite Mk V] Increase the reset delay for the touchscreen               ║
║ • Account for late OCM loading                                               ║
║                                                                              ║
║ LabTop (bios) e tutti i dispositivi collegati potrebbero non essere          ║
║ utilizzabili durante l'aggiornamento.                                        ║
╚══════════════════════════════════════════════════════════════════════════════╝
Eseguire l'operazione? [Y|n]: y
Scaricamento…            [*                                      ] Manca meno diScaricamento…            [*                                      ] Manca meno diScaricamento…            [**                                     ] Manca meno diScaricamento…            [***                                    ] Manca meno diScaricamento…            [***                                    ] Manca meno diScaricamento…            [****                                   ] Manca meno diScaricamento…            [****                                   ] Manca meno diScaricamento…            [*****                                  ] Manca meno diScaricamento…            [*****                                  ] Manca meno diScaricamento…            [******                                 ] Manca meno diScaricamento…            [******                                 ] Manca meno diScaricamento…            [*******                                ] Manca meno diScaricamento…            [*******                                ] Manca meno diScaricamento…            [********                               ] Manca meno diScaricamento…            [********                               ] Manca meno diScrittura…               [                                       ]
invalid image size 0x800000, expected 0x1000000`

soc7, 2025-01-13:
I was able to upgrade to latest coreboot version  using: $ sudo flashrom -p internal -w 24.10.rom -i bios --ifd -N
and still the problem persists as before, no sound on audiojack, same headset that was working with AMI.

Sean-StarLabs, 2025-01-13:
Please provide the `dmesg`

Sean-StarLabs, 2025-02-24:
[coreboot.zip](https://github.com/user-attachments/files/18944991/coreboot.zip)

Can you try this build please?

soc7, 2025-02-24:
$ sudo flashrom -p internal -w coreboot.rom -i bios --ifd -N
flashrom unknown on Linux 6.1.0-31-amd64 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
ERROR: Could not get I/O privileges (Operation not permitted).
Make sure you are root. If you are root, your kernel may still
prevent access based on security policies.
On OpenBSD set securelevel=-1 in /etc/rc.securelevel and
reboot, or reboot into single user mode.
On NetBSD reboot into single user mode or make sure
that your kernel configuration has the option INSECURE enabled.
Error: Programmer initialization failed.

Sean-StarLabs, 2025-02-24:
Sorry - no idea, never used BSD! Sounds like the equivalant of iomem (solution on linux is to set to relaxed) or kernel lockdown (disable  secureboot)

soc7, 2025-02-24:
It is debian bookworm

Sean-StarLabs, 2025-02-24:
Then it'll be one of those two solutions

soc7, 2025-02-24:
Indeed, just installed it and how headphone jack works...I've bought a new dauther board thinking it was the problem. Thanks anyway, good to have it finally working.

closed 2025-02-25 by Sean-StarLabs

soc7, 2025-02-25:
The battery is apparently lasting much longer on this release, amazing work. Thanks Sean