StarLabsLtd/firmware #244: Error when installing system updates grub-efi-amd64

rogercreagh, 2025-01-20:
StarLite V with firmware version 24.10 and Ubuntu 24.04.1

Ubuntu system updates has just stopped with the following message:

[image]

The help message just says "The grub boot loader was previously installed to a disc which is no longer present, or whose unique identifier has changed for some reason. It is important to make sure the installed GRUB core image stays in sync with GRUM modules and grub.cfg. Please check again to make sure GRUB is written to the appropriate boot devices."

If I check the box it then warns me that grub will not be installed and it won't boot, if I leave the box unchecked it warns me that grub may clash with the installed modules and not boot.

Anyone any idea what on earth I should do. Is this a coreboot problem or an ubuntu one?

Anachron, 2025-01-20:
It's something specific to your setup. Nothing related to this repository at least.

You can run the command with `sudo` or as `root` and see if it works there. 

Did you verify if your disk is still accessible and the UUID of it is the same?

rogercreagh, 2025-01-20:
Yep thanks, everything working because the update hasn't completed yet. The disc UUID is the same as is specified in the current grub.cfg. There is no indication as to what cmd failed - the usual raft of things were getting updates and the process has halted about 80% through.
Not a problem for here I think but worth being aware if you are doing software updates on a StarliteV Ubuntu24.04 - have a boot ubunutu usb stick just in case...

closed 2025-01-20 by rogercreagh