StarLabsLtd/firmware #245: Sandisk 2TB microsd card not usable with Starlabs Starlite tablet

blissd, 2025-01-21:
### **Describe the bug**
I've installed a Sandisk 2TB microsdxc card in my Starlabs Starlite tablet, but the tablet is unable to use the card. Through the GNOME Disks app the card is not visible for mounting for formatting. Running `dmesg` shows the following error message repeated many times:

```
[ 1384.903660] mmc0: Card stuck being busy! __mmc_poll_for_busy
[ 1384.903677] mmc0: error -110 whilst initialising SD card
```

---

### **Steps to Reproduce**
Detailed steps to reproduce the issue:
1. Install a 2TB Sandisk microsd card.
2. Try to use it.
---

### **Expected Behavior**
I expect the card to be usable by the tablet.

---

### **Screenshots**
n/a

---

### **System Information**
Provide your system details:
- **OS:** Fedora Silverblue 41
- **Kernel Version:**  6.12.9-200.fc41.x86_64
- **Product SKU:** I5
- **BIOS Version:** 24.10
- **EC Release:** 24.10

---

### **Logs**
Include any relevant logs that might help diagnose the issue:
- **Dmesg Output:**
```
[   50.541454] mmc0: Card stuck being busy! __mmc_poll_for_busy
[   50.541480] mmc0: error -110 whilst initialising SD card
```

---

### **Additional Context**
Provide any other information or context that might help in debugging this issue. This could include:
- Previous steps taken to resolve the issue. None.
- Workarounds attempted, if any. None.
- Frequency of the issue: Always

---

Sean-StarLabs, 2025-01-21:
Basically the same as #183.  Until patches are merged and released upstream, a workaround would be to use the older rts5139  driver, either from linux git or one of the many forks (i.e. [this one](https://github.com/asymingt/rts5139))

Sean-StarLabs, 2025-01-24:
linux commit 0d8521dcd236d1b8f664f54a0309e96bfdcb4f9.

The staging driver that was removed, `rts5139`, worked well with multiple
Realtek card readers and provided more comprehensive functionality than
`rtsx_usb`. It supported features like interrupts, SSC, speed modes,
suspend, secure erase, TRIM, and more. Additionally, its error recovery
mechanisms were superior.

Need to add these to rtsx_usb

Sean-StarLabs, 2025-11-11:
https://github.com/torvalds/linux/compare/master...StarLabsLtd:linux:wip/sean/rtsx