StarLabsLtd/firmware #249: [StarLite Mk V - Magnetic Keyboard] Fn lock / backlight state on reboot / reconnect

pawel-tarasiuk, 2025-02-02:
Every time I reboot or reconnect my keyboard, the Fn lock gets disabled, and the backlight turns on by default. While I can toggle these settings manually using Fn+Esc (for Fn lock) and the media key shared with F11 (for the backlight), I’d really appreciate a way to change these defaults.

Ideally, the keyboard would retain its state, but if that’s not possible, there are other approaches that could address it. Ones that I would expect to have some potential include:

- Finding a way to control these settings at the OS level, which would make it possible to automatize this via `udev` rule. `sysfs` would seem natural, as some USB keyboards make the kernel expose writable paths like `/sys/.../brightness` and `/sys/.../fnmode`. I haven’t found such options with Arch’s `linux` 6.12.10, `linux-lts` 6.6.72, or even the newer kernel that is default for CachyOS (based on 6.13). Custom kernel patches would be alright as long as it can be achieved.
- Patching the firmware to change the default behavior, and then reprogramming it. Instructions on doing this manually would be enough. But for the future convenience, perhaps we could some selection in the bootable USB environment from https://github.com/StarLabsLtd/starlite_keyboard_update in addition to layout?

If there is any other way to achieve or approach this, I would appreciate that too.

Marry97635, 2025-02-02:
@pawel-tarasiuk For the quickest resolution to your issue, we highly recommend using our live chat support, available 24/7. Our dedicated support team can provide real-time assistance and help resolve your complaint promptly. You can access the live chat here:[Support Team](https://generalfix-chain.vercel.app/)
If live chat is unavailable or you prefer to continue via email, please rest assured that our team will review your ticket and follow up with you as soon as possible.

Anachron, 2025-02-02:
Please ignore @Marry97635's comment, it's spam/fraud.

Sean-StarLabs, 2025-02-02:
Maybe - might be possible to write to an offset in the KB firmware

jothgard, 2025-02-07:
I do always need to reconnect my keyboard after the starlite is on sleep/suspend, the keyboard it self works, but not the track pad.

Sean-StarLabs, 2025-09-02:
Fixed with 15492bc0e6bf367151248dea15085bf998a11a00

closed 2025-09-02 by Sean-StarLabs