StarLabsLtd/firmware #251: [Starbook MK V / MK VI] Battery set to 60/80 percent surpasses the limit when device turned off

kostadinnm, 2025-02-26:
### **When turning off a Starbook and there's a threshold set to the battery level in bios/uefi, the charging continues**
As a result, when back on, the device continues to charge up to 100%. If already fully charged, it doesn't go down to the target value.

---

### **Steps to Reproduce**
Detailed steps to reproduce the issue:
1. Open bios/uefi
2. Go to battery
3. Set max charge level to 60% or 80%

---

### **Expected Behavior**
The battery shouldn't continue to charge after shutting down

---
### **System Information**
- **OS:** EndeavourOS Linux x86_64
- **Kernel Version:** 6.12.16-1-lts
- **Product SKU:** StarBook MK VI (AMD Ryzen 7 5800U)
- **BIOS Version:** UEFI 1.16.0
- **EC Release:** 2.0

---

### **Additional Context**
- Previous steps taken to resolve the issue: update firmware
- Frequency of the issue: "Always"

---

As a side note: believing in the multi-purposeness of USB-C, I ordered mine with the USB-C charger instead if the regular one. That was a twofold mistake:

- personal and purely "usability" one, where I sometimes need the UCB-C port or side for second purpose(i.e. monitor, second usb-c-interfaced device, desk placement, etc.)
- the charging experience in itself is strange. Strange by means that I feel some electrical current flowing trough the device body panels. I suspect the USB-C port isn't grounded properly. Also the charger had intermittent refusals to re-engage charging. I am talking about this "power efficient"(**GaN**), but "extremely value" engineered DC adapter: [Starlabs 65W GaN USB-C Power Adapter](https://bg.starlabs.systems/collections/power-cables/products/65w-gan-usb-c-power-adapter), which is an electronic waste of mine already.

The experiences with the laptop I've had, along with the observations from the second point above and the apparent similarity of many AMD-based models that come from the Chinese manufacturer, formed my **_personal opinion_** that there had been nothing special(remember - chassis design, special alloy, etc..) that the brand Starlabs provided me back in 2022/2023(don't remember). Add to that the wait time, which should've fine-tuned the pure hardware stuff(like charging, fan noise, in-spec-but-unpleasant temps on light loads, rubber feet falling, fingerprint's no-go, etc.) and the decline you're facing is unavoidable, imho. It seems to me that you've accepted everything the manufacturer gave you with little to no iterations on Quality Control. The Starfighter model should've been a masterpiece to counter all things I mentioned, but I don't want comment it at all. Anyways, this issue is for the charging threshold logic and I expect some discussion on that.

I don't have high hopes for this, but to the least me, as an author, should be able to post comments, if you decide to close the issue unresolved, right?!

Happy computing!

Sean-StarLabs, 2025-02-26:
This is expected; the EC doesn't have any storage, so every time it starts, it starts with default values (charge to 100%, 1.0C speed, etc.). The way it's configuriable in the firmware (which we do so you can still configure it with any OS), basically "sets" it in the EC the firmware starts.

i.e. StarBook is entirely off (G3), you connect a charger, the EC starts with its default values, so it'll charge to 100%, etc. When you turn it on, your settings will be applied, say charging limited to 80% or whatever (it will discharge to 80%, but that takes quite a while if you have a charger connected). That does mean that if you turn off with the charger connected and leave it, that 80% preference will remain intact.

In my very limited spare time, I've been playing around with a way to get the EC to read the settings from the firmware rather than the firmware writing to the EC. Should I have any luck, that'll all change, but for now, it is a "won't fix" because we want to maintain having the options in the firmware so as not to create a dependency for the OS.

For the chargers, the ones we use are not grounded; you can use a grounded charger if you want to avoid this.

As for your feedback, it's always best to ping us an email - after all, all of our hardware is literally the product of feedback. As for the "nothing special", we definitely haven't "accepted" anything; the StarBook is on its 37th iteration, and we are a complete pain for a factory for tweaking and adjusting things. For example, with the fingerprints, for the last  2 years or so, we've asked our customers, do we switch from black to save fingerprints or just stick with it? The feedback finally tipped last year, so now the finish and colour are different.

closed 2025-02-26 by Sean-StarLabs

kostadinnm, 2025-02-26:
Pardon my incompetence, by what does "EC" actually mean?!
From a usability perspective, when the laptop gets "cold"-started, what prevents it from deciding whether to charge or not when the reading of the charge_level and target_charge_level values are present?! Whatever the component schematic is, you should be able to tell it two things dynamically: engage charging, disengage charging.
I won't mind dependency(and transparency for that matter) on a custom package for the major linux distros to get battery charging limit(and maybe others) and if it's communicated well(i.e. if you make it clear we're all loosing this if the package is not installed). Getting half-baked feature, saying you're working on it, but won't share the progress is hypocritical from your side.

The charger need not have a third connector point for grounding - many don't. It's the usb-c port on the laptop that I suspect leaking current. Maybe this is due to the board in my laptop is AMD- based and both you and the Chinese you source it from are negligible on perfecting those.

Fingerprint is a small issue and I guess few people use it that much on their laptops, but even tinier concern is the fingerprint's colour flavour. I mean, it shouldn't break the whole design aesthetic of the chassis (and I prefer blacks), but first and foremost it should work. So I don't consider settling on the colour a significant effort neither from you, nor from the manufacturer. On the Starbook VI for example, it's a bit odd that it's the same color as it surrounding and that is on the farthest corner to my right-hand, but it's raised and the laptop is compact so I can easily find it each time I reach for it.

I am writing all this without even a glimpse of self-censoring and my intent is not nagging pointlessly with you, but am really curious on the matter of mobile computing. I wish you good times communicating your point victories and maybe full success stories.

Sean-StarLabs, 2025-02-27:
> Pardon my incompetence, by what does "EC" actually mean?!

https://en.wikipedia.org/wiki/Embedded_controller#:~:text=An%20Embedded%20Controller%20(EC)%20is,platforms%20(such%20as%20laptop).

> what prevents it from deciding whether to charge or not when the reading of the charge_level and target_charge_level values are present?!

Coming up from G3, it'll just use the default values (i.e. 100%) because nothing has told it otherwise.

> I won't mind dependency(and transparency for that matter) on a custom package for the major linux distros to get battery charging limit(and maybe others) and if it's communicated well(i.e. if you make it clear we're all loosing this if the package is not installed).

You're probably not alone, but it doesn't overcome the technical and logistic hurdles of creating a utility that'll run on every OS, and there are many.

> Getting half-baked feature

Just because it's not how you want it, doesn't mean its wrong :)

> saying you're working on it, but won't share the progress is hypocritical from your side.

I do not understand how that's hypocritical, if you want to know something, ask. But for progress, it can't be quantified - I've just been trying to get the initialised values of BRAM into a consistent state which would be the first hurdle.