StarLabsLtd/firmware #260: [StarBook MkV] Resume From Suspend in Coreboot 25.02 Broken?

s3ph-scott, 2025-03-11:
### **Describe the bug**
I hate to jump on things so quickly however I've upgraded my StarBook MKV to Coreboot 25.02 from LVFS this morning and noted an issue with resuming from suspend in Ubuntu 24.04 with HWE kernel 6.11. Instead of waking up from Suspend to the login screen I get a very broken login and eventually terminal output logging messages to do with reads and writes to what I understand to be log files. This process appears to have been somehow broken to what looks like issues with reading & writing to the SSD. Of note is that my laptop is using LUKS encryption on the disk.

---

### **Steps to Reproduce**
Detailed steps to reproduce the issue:

1. Upgrade the StarBook MkV to Coreboot 25.02.
2. Send the laptop to Suspend.
3. Wake the laptop up via keyboard press/mouse shake. The unlock screen appears but note at this point the mouse cursor is represented by a white square and characters do not type into the password field.
4. Eventually the system outputs the terminal where error messages from systemd are being logged. 

---

### **Expected Behaviour**
The laptop should resume from Suspend and allow the user to login once again.

---

### **Screenshots**
This is the eventual terminal output after attempting to login with the password to unlock:

---

### **System Information**
Provide your system details:
- **OS:**  Ubuntu 24.04
- **Kernel Version:** 6.11.0-19-generic
- **Product SKU:** B5
- **BIOS Version:** 25.02
- **EC Release:** 24.10

---

### **Logs**
- **Dmesg Output:** I have attached a screenshot that happens during the issue. As I think there's a failure to read/write the disk I am not sure if dmesg is logging the error.

---

### **Additional Context**
I was eager to try Coreboot 25.02 as there is an improvement for waking the laptop using the keyboard. I have noted in previous versions of Coreboot that if the laptop has been suspended for a while and is left connected to a Phillips 346P monitor via USB-C which charges the laptop it won't reliably come back from suspend. Instead will remain on a blank screen and needs a long press of the power key to reboot. Whether that's similar to the issue specified above though I'm not certain.

---

Sean-StarLabs, 2025-03-11:
How many times has it happened? If more than once, can you try it with clock gating disabled?

s3ph-scott, 2025-03-17:
I have performed a reinstall of the 25.02 firmware today in order to correct the discrepancy between the Coreboot and EC versions. Unfortunately the issue persists.

I've also performed the Coreboot 25.02 update on a Starlite MKV that runs the exact same Ubuntu release and Kernel. This does not have an issue with Suspend.

robbertzzz, 2025-04-01:
@Sean-StarLabs advised my via email that I have this issue, although it presents itself differently on my device so I just wanted to log my experience here as well. When trying to wake from suspend either with keyboard or mouse, the LEDs with a light bulb icon (one above the keyboard, one on the left side) both start flashing rapidly. My screen does not wake up, there are no sounds or anything, just these two LEDs going mad. Same Coreboot version (25.02), on Ubuntu 24.04.1 LTS, no disk encryption. I can only long press the power button to restart, which sounds similar to what @s3ph-scott described happens with an external display although for me it happens without any peripherals connected.

Sean-StarLabs, 2025-04-01:
@robbertzzz Not related - different board

Sean-StarLabs, 2025-04-08:
Fixed with https://review.coreboot.org/c/coreboot/+/87213

closed 2025-04-08 by Sean-StarLabs