StarLabsLtd/firmware #266: [StarBook Mk V] Failed UEFI dbx update

vanntile, 2025-03-17:
### **Describe the bug**
Failed to update firmware to UEFI dbx 20241101

### **Steps to Reproduce**
Run the firmware update.

```sh
sudo fwupdmgr update --force

╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade UEFI dbx from 421 to 20241101?                                       ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This updates the list of forbidden signatures (the "dbx") to the latest      ║
║ release from Microsoft.                                                      ║
║                                                                              ║
║ An insecure version of Howyar's SysReturn software was added, due to a       ║
║ security vulnerability that allowed an attacker to bypass UEFI Secure Boot.  ║
║                                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: Y
Writing…                 [                                       ]
failed to write data to efivarfs: Error writing to file descriptor: No space left on device
```

---

### **System Information**
Provide your system details:
- **OS:** Pop!_OS 22.04 LTS
- **Kernel Version:** Linux 6.12.10-76061203-generic
- **Product SKU:** B5
- **BIOS Version:** 24.10
- **EC Release:** 1.3

---

Any way to increase the mount size? is the UEFI NVRAM fully mounted? How would one go forward from here

Anachron, 2025-03-18:
How much space do you have (left) on your EFI partition?

`LC_ALL=C df -h | grep -iE 'Filesystem|/boot/efi'` on my setup returns

```
Filesystem            Size  Used Avail Use% Mounted on
/dev/nvme0n1p1        300M  152K  300M   1% /boot/efi
```

vanntile, 2025-03-18:
Is that actually the correct mount we are looking for? I have

```
LC_ALL=C df -h | grep -iE 'Filesystem|/boot/efi'
Filesystem             Size  Used Avail Use% Mounted on
/dev/nvme0n1p1         497M  286M  212M  58% /boot/efi
```

but the one I think is relevant:

```
LC_ALL=C df -h | grep -iE 'Filesystem|/sys/firmware/efi/efivars'
Filesystem             Size  Used Avail Use% Mounted on
efivarfs                64K   54K  5.1K  92% /sys/firmware/efi/efivars
```

Sean-StarLabs, 2025-03-18:
We'll include the dbx in the next release, which is about a week or so away so might be easier to wait until then

vanntile, 2025-03-18:
Ok, thank you, will close the issue then.

Sean-StarLabs, 2025-04-02:
Fixed with https://github.com/StarLabsLtd/edk2/commit/6ba1c5f5e3b6c8ba99b6ed188afece24e76b983b

closed 2025-04-02 by Sean-StarLabs

Sean-StarLabs, 2025-08-02:
Sorry, not really sure what you're asking - it's something that's updated regularly, and we put the latest one in each time we do a release.

vanntile, 2025-08-02:
No problem, I can be more explicit. Yes, I understand you update the dbx version in each release. But, as you can see, even though you have updated it, using fwupdmgr to update firmware on my Starbook keeps failing to update the dbx, for the same cause (the efivarfs size is insufficient to rewrite the dbx file, as only about 20kb are available and supposedly the new files are over 24kb). Do you have any instructions on how one would be able to remove the dbx file manually such that the fwupdmgr would succeed in updating to the latest release? Would that even be safe? Is there any solution?

When this same issue appeared for other brands, the final verdict in the cases where the machines were not bricked was to get the OEM to push an improved update, so that's what I'm looking for.

vanntile, 2025-08-02:
> You can remove the dbx; wouldn't affect what fwupd is trying to do.

Any help with that? Although using lsattr doesn't show it is an immutable file, I don't know how to remove it

> Just wait for the next coreboot update.

Ok, will do, thank you

Sean-StarLabs, 2025-08-03:
> Any help with that? 

Change the mode from Standard to Custom, then you'll have an option underneath to delete or enroll keys