StarLabsLtd/firmware #267: 25.03 Release

Sean-StarLabs, 2025-03-18:
Changes:
- [ ] Closes #266

Anachron, 2025-03-25:
Hey Sean, how do I install the `25.03` or even the `25.02` firmware release for the `StarLite MK V`? 

There is no `25.02.rom` Inside https://github.com/StarLabsLtd/firmware/tree/main/StarLite/MkV/coreboot/25.02, yet `I5.coreboot.metainfo.xml` refers to it.

`25.03` would be `I5-SB` here, but there is nothing inside `daily_builds` with such a name too, so I am not sure how I am getting the fix for "[StarLite Mk V] Inconsistent charger detection is back #265".

Sean-StarLabs, 2025-03-25:
> Hey Sean, how do I install the 25.03 or even the 25.02 firmware release for the StarLite MK V?

https://support.starlabs.systems/kb/firmware/getting-started

> There is no 25.02.rom Inside https://github.com/StarLabsLtd/firmware/tree/main/StarLite/MkV/coreboot/25.02, yet I5.coreboot.metainfo.xml refers to it.

There is, but as always, not recommend - especially for those without programmers.

> nothing inside daily_builds with such a name

```
These are available for:

* StarBook Mk VII Ultra
```

> getting the fix for "[StarLite Mk V] Inconsistent charger detection is back https://github.com/StarLabsLtd/firmware/issues/265".

#265 is an open issue, there isn't a fix for it yet

closed 2025-04-29 by Sean-StarLabs