StarLabsLtd/firmware #268: [StarLite Mk V] Instability after upgrading to 25.02

owenthomas17, 2025-03-22:
### **Describe the bug**
Since upgrading to the new 25.02 firmware the other day, when my StarLite is plugged into power, a few minutes after booting and logging into the system it completely powers itself off instantly. Normally there are problems with bluetooth/wireless chip as well when plugged into power. However when running from battery it seems to at least keep working, enough to write this bug report! Possibly this is a hardware error and unfortunate timing with the upgrade. But the timing did point me back to that event.

---

### **Steps to Reproduce**
Detailed steps to reproduce the issue:
1. Plug in power cable.
2. Boot the machine and log in.
3. Machine powers off after a few minutes.

Be as specific as possible to help us replicate the issue.

---

### **Expected Behavior**
Well, um, not to fail hard enough to power down :) 

---

### **System Information**
Provide your system details:
- **OS:** Ubuntu 24.04.2
- **Kernel Version:** 6.11.0-19-generic
- **Product SKU:** I5
- **BIOS Version:** 25.02
- **EC Release:** 25.2

---

### **Logs**
Include any relevant logs that might help diagnose the issue:
- **Dmesg Output:** See attached dmesg logs.

[dmesg-star-lite-V.log](https://github.com/user-attachments/files/19403584/dmesg-star-lite-V.log)

The specific lines pointing me to the embedded controller

```bash
[    5.146811] ACPI Error: Timeout from EC hardware or EC device driver (20240322/evregion-311)

[    5.147332]
               Initialized Local Variables for Method [ECRD]:
[    5.147334]   Local0: (____ptrval____) <Obj>           Integer 0000000000000000

[    5.147339] Initialized Arguments for Method [ECRD]:  (1 arguments defined for method invocation)
[    5.147340]   Arg0:   (____ptrval____) [RefOf] <Node>          Name LSTE RegionField (____ptrval____)

[    5.147347] ACPI Error: Aborting method \_SB.PCI0.LPCB.EC.ECRD due to previous error (AE_TIME) (20240322/psparse-529)
[    5.147353] ACPI Error: Aborting method \_SB.PCI0.LPCB.EC._REG due to previous error (AE_TIME) (20240322/psparse-529)
[    5.147359] ACPI: EC: EC_CMD/EC_SC=0x66, EC_DATA=0x62
[    5.147361] ACPI: \_SB_.PCI0.LPCB.EC__: Boot DSDT EC used to handle transactions
[    5.147363] ACPI: Interpreter enabled
[    5.147375] ACPI: PM: (supports S0 S3 S4 S5)
[    5.147377] ACPI: Using IOAPIC for interrupt routing
[    5.147615] PCI: Using host bridge windows from ACPI; if necessary, use "pci=nocrs" and report a bug
[    5.147617] PCI: Ignoring E820 reservations for host bridge windows
[    5.147810] ACPI: Enabled 3 GPEs in block 00 to 7F
[   10.148260] ACPI Error: AE_TIME, Returned by Handler for [EmbeddedControl] (20240322/evregion-301)
[   10.148263] ACPI Error: Timeout from EC hardware or EC device driver (20240322/evregion-311)

[   10.148333]
               Initialized Local Variables for Method [ECRD]:
[   10.148334]   Local0: (____ptrval____) <Obj>           Integer 0000000000000000

[   10.148339] Initialized Arguments for Method [ECRD]:  (1 arguments defined for method invocation)
[   10.148340]   Arg0:   (____ptrval____) [RefOf] <Node>          Name ECPS RegionField (____ptrval____)

[   10.148348] ACPI Error: Aborting method \_SB.PCI0.LPCB.EC.ECRD due to previous error (AE_TIME) (20240322/psparse-529)
[   10.148354] ACPI Error: Aborting method \_SB.PCI0.LPCB.EC.BAT0._STA due to previous error (AE_TIME) (20240322/psparse-529)
```
---

### **Additional Context**
Provide any other information or context that might help in debugging this issue. This could include:
- Previous steps taken to resolve the issue: Seems to run so far on Battery.
- Workarounds attempted, if any: N/A
- Frequency of the issue: Always when running from power.
---

owenthomas17, 2025-03-24:
I did some more testing on this over the weekend, I managed to get a stable enough boot that I was able to actually shut the machine down cleanly. I'd read on the forum that in some cases removing the power once shutdown so I did that and left a few minutes. On the next boot I don't seem to be able to reproduce the same symptoms. I'll leave the issue open for now, but feel free to close as I'mmo no longer having and issue. If you'd like any more diagnostics happy to provide them.

Sean-StarLabs, 2025-03-24:
> I'd read on the forum that in some cases removing the power once shutdown so I did that and left a few minutes

That lets the EC update, so it would make sense. I'll close now, but let me know if you see it again!

closed 2025-03-24 by Sean-StarLabs

dchang0, 2025-04-07:
I have something similar. Since 25.02, I've been seeing random shutdowns. Sometimes it was when I was away from the computer, sometimes when using it. They accelerated in frequency until this past Saturday night (4/5/2025), when my StarLite Mk V was shutting down maybe once an hour.

Now I can't even boot to get my files off before sending the device in for warranty repair...

In my case, It doesn't appear to be reaching the POST stage. The power light turns solid white and the charging circuit works correctly (charges to full, will discharge to empty if left on for 6-8 hours). But the screen stays dark, no boot logo ever comes up, and the keyboard cover backlight doesn't work.

I suspect the BIOS got corrupted but not sure. Could also be a hardware issue (loose display cable? does this prevent boot?)

Two things are for sure: 1) Nothing like this happened under firmware 24.xx, and 2) it's not the OS--it never gets far enough to boot the OS.

Please let me know if you have any tips on how to resurrect my tablet--thanks!

Sean-StarLabs, 2025-04-08:
> I have something similar. 

OP's issue was EC driver timing out - that's not going to have any bearing on the display.

Given the drain, it sounds like yours is on, but no display output - be worth connecting a monitor if you have one, but either way, it's one for support.