StarLabsLtd/firmware #269: [StarLite Mk V] Do not yet upgrade to linux 6.14, breaks wireless

BlackIkeEagle, 2025-03-25:
### **Describe the bug**
When you upgrade to linux 6.14 something goes seriously wrong with iwlwifi, further research needed for details

---

### **Steps to Reproduce**
Detailed steps to reproduce the issue:
1. upgrade to linux 6.14
2. reboot
3. no more wifi

---

### **System Information**
Provide your system details:
- **OS:** Arch Linux
- **Kernel Version:** 6.14.0
- **Product SKU:** [Output of `cat /sys/class/dmi/id/product_sku`]
- **BIOS Version:** [Output of `cat /sys/class/dmi/id/bios_version`]
- **EC Release:** [Output of `cat /sys/class/dmi/id/ec_firmware_release`]

---

### **Logs**
Include any relevant logs that might help diagnose the issue:
- **Dmesg Output:** Attach or paste the relevant portion of `dmesg` logs.
- **Other Logs:** Specify and include logs from related subsystems if applicable.

Anachron, 2025-03-25:
Can you please at least post your `dmesg` so that I know what to look for? 
I am not too keen on avoiding newer kernel releases.

BlackIkeEagle, 2025-03-25:
Yeah I'm collecting the info and will update

BlackIkeEagle, 2025-03-25:
`modinfo iwlwifi | wl-copy`: note this is from another machine, not the starlite

```
filename:       /lib/modules/6.14.0-1-BEDE/kernel/drivers/net/wireless/intel/iwlwifi/iwlwifi.ko.zst
description:    Intel(R) Wireless WiFi driver for Linux
license:        GPL
firmware:       iwlwifi-1000-5.ucode
firmware:       iwlwifi-100-5.ucode
firmware:       iwlwifi-2000-6.ucode
firmware:       iwlwifi-2030-6.ucode
firmware:       iwlwifi-105-6.ucode
firmware:       iwlwifi-135-6.ucode
firmware:       iwlwifi-5000-5.ucode
firmware:       iwlwifi-5150-2.ucode
firmware:       iwlwifi-6000-6.ucode
firmware:       iwlwifi-6050-5.ucode
firmware:       iwlwifi-6000g2a-6.ucode
firmware:       iwlwifi-6000g2b-6.ucode
firmware:       iwlwifi-7260-17.ucode
firmware:       iwlwifi-3160-17.ucode
firmware:       iwlwifi-3168-29.ucode
firmware:       iwlwifi-7265-17.ucode
firmware:       iwlwifi-7265D-29.ucode
firmware:       iwlwifi-8000C-36.ucode
firmware:       iwlwifi-8265-36.ucode
firmware:       iwlwifi-9000-pu-b0-jf-b0-46.ucode
firmware:       iwlwifi-9260-th-b0-jf-b0-46.ucode
firmware:       iwlwifi-Qu-b0-hr-b0-77.ucode
firmware:       iwlwifi-Qu-c0-hr-b0-77.ucode
firmware:       iwlwifi-Qu-b0-jf-b0-77.ucode
firmware:       iwlwifi-QuZ-a0-hr-b0-77.ucode
firmware:       iwlwifi-QuZ-a0-jf-b0-77.ucode
firmware:       iwlwifi-cc-a0-77.ucode
firmware:       iwlwifi-so-a0-jf-b0-89.ucode
firmware:       iwlwifi-so-a0-hr-b0-89.ucode
firmware:       iwlwifi-so-a0-gf-a0-89.ucode
firmware:       iwlwifi-so-a0-gf-a0.pnvm
firmware:       iwlwifi-ty-a0-gf-a0-89.ucode
firmware:       iwlwifi-ty-a0-gf-a0.pnvm
firmware:       iwlwifi-ma-a0-hr-b0-89.ucode
firmware:       iwlwifi-ma-a0-gf-a0-89.ucode
firmware:       iwlwifi-ma-a0-gf-a0.pnvm
firmware:       iwlwifi-ma-a0-gf4-a0-89.ucode
firmware:       iwlwifi-ma-a0-gf4-a0.pnvm
firmware:       iwlwifi-ma-b0-hr-b0-89.ucode
firmware:       iwlwifi-ma-b0-gf-a0-89.ucode
firmware:       iwlwifi-ma-b0-gf-a0.pnvm
firmware:       iwlwifi-ma-b0-gf4-a0-89.ucode
firmware:       iwlwifi-ma-b0-gf4-a0.pnvm
firmware:       iwlwifi-bz-a0-hr-b0-96.ucode
firmware:       iwlwifi-bz-a0-gf-a0-96.ucode
firmware:       iwlwifi-bz-a0-gf-a0.pnvm
firmware:       iwlwifi-bz-a0-gf4-a0-96.ucode
firmware:       iwlwifi-bz-a0-gf4-a0.pnvm
firmware:       iwlwifi-bz-a0-fm-b0-96.ucode
firmware:       iwlwifi-bz-a0-fm-b0.pnvm
firmware:       iwlwifi-bz-a0-fm-c0-96.ucode
firmware:       iwlwifi-bz-a0-fm-c0.pnvm
firmware:       iwlwifi-bz-a0-fm4-b0-96.ucode
firmware:       iwlwifi-bz-a0-fm4-b0.pnvm
firmware:       iwlwifi-gl-b0-fm-b0-96.ucode
firmware:       iwlwifi-gl-b0-fm-b0.pnvm
firmware:       iwlwifi-gl-c0-fm-c0-96.ucode
firmware:       iwlwifi-gl-c0-fm-c0.pnvm
firmware:       iwlwifi-sc-a0-fm-b0-96.ucode
firmware:       iwlwifi-sc-a0-fm-b0.pnvm
firmware:       iwlwifi-sc-a0-fm-c0-96.ucode
firmware:       iwlwifi-sc-a0-fm-c0.pnvm
firmware:       iwlwifi-sc-a0-hr-b0-96.ucode
firmware:       iwlwifi-sc-a0-hr-b0-96.ucode
[... 487 lines cut ...]
		22:2D:5B:03:3E:4A:57:1E:97:4C:41:49:23:C6:FA:7A:44:2F:DA:F0:
		F0:A4:8F:13:69:57:F2:7E:32:6F:9C:A2:CA:6C:EC:DE:E6:68:68:86:
		67:3A:B8:A7:F6:8F:B7:40:12:AB:30:CD:64:86:F5:F6:EA:79:CB:FB:
		6D:81:B0:3C:F7:2F:B4:84:DE:F2:AA:5D:E5:70:3E:C5:10:B1:89:66:
		FF:D1:99:1D:5F:11:03:16:A4:13:10:EF:C8:B9:93:3B:00:03:D9:E5:
		6B:4C:21:82:81:D9:EF:2F:B8:60:3E:60:91:79:89:11:8F:74:AF:44:
		66:50:DE:E7:5C:78:D9:8D:F6:6A:F2:EB:15:92:27:34:BE:8B:CF:52:
		B0:B4:26:63:4C:35:2B:22:41:97:3B:9F:F6:5F:B2:15:FC:D8:21:2C:
		DB:18:8E:EF:BE:F4:1F:DD:14:BE:CC:34
parm:           disable_11be:Disable EHT capabilities (default: false) (bool)
parm:           disable_11ax:Disable HE capabilities (default: false) (bool)
parm:           remove_when_gone:Remove dev from PCIe bus if it is deemed inaccessible (default: false) (bool)
parm:           disable_11ac:Disable VHT capabilities (default: false) (bool)
parm:           power_level:default power save level (range from 1 - 5, default: 1) (int)
parm:           power_save:enable WiFi power management (default: disable) (bool)
parm:           led_mode:0=system default, 1=On(RF On)/Off(RF Off), 2=blinking, 3=Off (default: 0) (int)
parm:           bt_coex_active:enable wifi/bt co-exist (default: enable) (bool)
parm:           enable_ini:0:disable, 1-15:FW_DBG_PRESET Values, 16:enabled without preset value defined,Debug INI TLV FW debug infrastructure (default: 16) (uint)
parm:           uapsd_disable:disable U-APSD functionality bitmap 1: BSS 2: P2P Client (default: 3) (uint)
parm:           nvm_file:NVM file name (charp)
parm:           fw_restart:restart firmware in case of error (default true) (bool)
parm:           amsdu_size:amsdu size 0: 12K for multi Rx queue devices, 2K for AX210 devices, 4K for other devices 1:4K 2:8K 3:12K (16K buffers) 4: 2K (default 0) (int)
parm:           11n_disable:disable 11n functionality, bitmap: 1: full, 2: disable agg TX, 4: disable agg RX, 8 enable agg TX (uint)
parm:           swcrypto:using crypto in software (default 0 [hardware]) (int)
```

BlackIkeEagle, 2025-03-25:
just for reference, just tested with linux-zen 6.14 and got some additional output because more debug flags are enabled there on iwlwifi

`sudo journalctl -b-1 | grep -i iwlwifi | wl-copy`

```
mrt 25 21:17:09 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: Detected crf-id 0x1300202, cnv-id 0x80400 wfpm id 0x80000030
mrt 25 21:17:09 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: PCI dev 54f0/0034, rev=0x370, rfid=0x105110
mrt 25 21:17:09 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: Detected Intel(R) Wireless-AC 9560 160MHz
mrt 25 21:17:09 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-jf-b0-89.ucode failed with error -2
mrt 25 21:17:09 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-jf-b0-88.ucode failed with error -2
mrt 25 21:17:09 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-jf-b0-87.ucode failed with error -2
mrt 25 21:17:09 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-jf-b0-86.ucode failed with error -2
mrt 25 21:17:09 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-jf-b0-85.ucode failed with error -2
mrt 25 21:17:09 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-jf-b0-84.ucode failed with error -2
mrt 25 21:17:09 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-jf-b0-83.ucode failed with error -2
mrt 25 21:17:09 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-jf-b0-82.ucode failed with error -2
mrt 25 21:17:09 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-jf-b0-81.ucode failed with error -2
mrt 25 21:17:09 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-jf-b0-80.ucode failed with error -2
mrt 25 21:17:09 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-jf-b0-79.ucode failed with error -2
mrt 25 21:17:09 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-so-a0-jf-b0-78.ucode failed with error -2
mrt 25 21:17:09 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: TLV_FW_FSEQ_VERSION: FSEQ Version: 0.0.2.36
mrt 25 21:17:09 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: loaded firmware version 77.f92b5fed.0 so-a0-jf-b0-77.ucode op_mode iwlmvm
mrt 25 21:17:10 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: WFPM_UMAC_PD_NOTIFICATION: 0x1f
mrt 25 21:17:10 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: WFPM_LMAC2_PD_NOTIFICATION: 0x1f
mrt 25 21:17:10 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: WFPM_AUTH_KEY_0: 0x90
mrt 25 21:17:10 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: CNVI_SCU_SEQ_DATA_DW9: 0x0
mrt 25 21:17:10 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: Detected RF JF, rfid=0x105110
mrt 25 21:17:10 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: RFIm is deactivated, reason = 4
mrt 25 21:17:10 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: base HW address: 8c:e9:ee:90:fb:fa
mrt 25 21:17:10 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: timeout waiting for FW reset ACK (inta_hw=0x0)
mrt 25 21:17:10 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: Start IWL Error Log Dump:
mrt 25 21:17:10 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: Transport status: 0x0000000A, valid: 6
mrt 25 21:17:10 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: Loaded firmware version: 77.f92b5fed.0 so-a0-jf-b0-77.ucode
mrt 25 21:17:10 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: 0x00000084 | NMI_INTERRUPT_UNKNOWN       
mrt 25 21:17:10 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: 0x00A082F0 | trm_hw_status0
mrt 25 21:17:10 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: 0x00000000 | trm_hw_status1
mrt 25 21:17:10 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: 0x004D2A06 | branchlink2
mrt 25 21:17:10 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: 0x004C92BE | interruptlink1
mrt 25 21:17:10 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: 0x004C92BE | interruptlink2
mrt 25 21:17:10 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: 0x00011F5A | data1
mrt 25 21:17:10 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: 0x01000000 | data2
mrt 25 21:17:10 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: 0x00000000 | data3
mrt 25 21:17:10 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: 0x00000000 | beacon time
mrt 25 21:17:10 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: 0x0001BFD1 | tsf low
mrt 25 21:17:10 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: 0x00000000 | tsf hi
mrt 25 21:17:10 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: 0x00000000 | time gp1
mrt 25 21:17:10 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: 0x0002763F | time gp2
mrt 25 21:17:10 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: 0x00000001 | uCode revision type
mrt 25 21:17:10 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: 0x0000004D | uCode version major
mrt 25 21:17:10 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: 0xF92B5FED | uCode version minor
mrt 25 21:17:10 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: 0x00000370 | hw version
mrt 25 21:17:10 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: 0x40480002 | board version
mrt 25 21:17:10 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: 0x802DFF00 | hcmd
mrt 25 21:17:10 archlinux-Yjk4NzBh kernel: iwlwifi 0000:00:14.3: 0x20020000 | isr0
[... cut ...]
mrt 25 21:17:14 archlinux-Yjk4NzBh kernel: RDX: 0000000000000000 RSI: 00007ffef06e5f00 RDI: 000000000000000d
mrt 25 21:17:14 archlinux-Yjk4NzBh kernel: RBP: 00007ffef06e5ea0 R08: 0000000000000000 R09: 0000000000000000
mrt 25 21:17:14 archlinux-Yjk4NzBh kernel: R10: 0000000000000000 R11: 0000000000000246 R12: 000057fe71e8a1a0
mrt 25 21:17:14 archlinux-Yjk4NzBh kernel: R13: 0000000000000018 R14: 00007ffef06e609c R15: 0000000000000000
mrt 25 21:17:14 archlinux-Yjk4NzBh kernel:  </TASK>
mrt 25 21:17:14 archlinux-Yjk4NzBh kernel: ---[ end trace 0000000000000000 ]---
```

Sean-StarLabs, 2025-03-26:
I think, from reading bits and pieces here and there, wifi is linux but for the TPM, is that with the ME on or off?

BlackIkeEagle, 2025-03-26:
dmesg of 6.14 attached

[linux-6.14-dmesg.txt](https://github.com/user-attachments/files/19471152/linux-6.14-dmesg.txt)

Sean-StarLabs, 2025-03-26:
That's with it masked? If so, can you post one with it out as all I can see there is that the Lite is correctly reporting no TPM

BlackIkeEagle, 2025-03-26:
Sorry, here the tpm2.service is no longer masked

BlackIkeEagle, 2025-03-26:
> That's with it masked? If so, can you post one with it out as all I can see there is that the Lite is correctly reporting no TPM

So you want dmesg with tpm2.service masked?

BlackIkeEagle, 2025-03-26:
New dmesg with tpm

[linux-6.14-dmesg-tpm2-unmasked.txt](https://github.com/user-attachments/files/19472264/linux-6.14-dmesg-tpm2-unmasked.txt)

BlackIkeEagle, 2025-03-26:
And the system does not hang persé, but it becomes very unresponsive while it it what feels like deadlocking on trying to get the wifi adapter up

Sean-StarLabs, 2025-03-27:
Erm, so, what's the issue with the TPM or is there no issue?

BlackIkeEagle, 2025-03-27:
There is no issue with the TPM, but wifi is completely broken when using linux 6.14.

I had tpm2.target masked since I installed this system before the tpm landed in the firmware and never unmasked it. I thought you alluded to a relation between the tpm2.target being masked and the wifi not working

Sean-StarLabs, 2025-03-27:
> There is no issue with the TPM, but wifi is completely broken when using linux 6.14.

Ah - got sent a screenshot of something Mastadon, must have been the wrong thing!

Can you try the daily build?

BlackIkeEagle, 2025-03-27:
Ah the prerelease of 25.03 has changes related to wifi and this missing `\_SB.PCI0.CNVW.CNVP._RST.PRRS` symbol?

Sean-StarLabs, 2025-03-27:
Yup

BlackIkeEagle, 2025-03-27:
It is an improvement in the sense that the system no longer feels clogged up and unresponsive, but still no working wifi. Might be another kernel bug?

[linux-6.14-dmesg-daily-fw.25.02-2025-03-27.txt](https://github.com/user-attachments/files/19492916/linux-6.14-dmesg-daily-fw.25.02-2025-03-27.txt)

Sean-StarLabs, 2025-03-27:
Thanks - can you post the early bit? It's not getting enabled from some reason.

> Might be another kernel bug?

Yes - given it works with the last 15 versions of the kernel, safe to assume something is wrong in 6.14 - but if we can work around it in fw, it's quicker and easier.

BlackIkeEagle, 2025-03-27:
> Thanks - can you post the early bit? It's not getting enabled from some reason.

I'm not sure what you mean with early bit, I could upload my journalctl from that boot if that gives you the information you need?

> 
> > Might be another kernel bug?
> 
> Yes - given it works with the last 15 versions of the kernel, safe to assume something is wrong in 6.14 - but if we can work around it in fw, it's quicker and easier.

Sean-StarLabs, 2025-03-27:
The dmesg you posted is from 123 seconds in - the problem is happening earlier, probably about 5 seconds in

BlackIkeEagle, 2025-03-27:
Ah I did not check that, excuse me

`sudo journalctl -b-2 --dmesg > linux-6.14-journalctl-daily-fw.25.02-2025-03-27.txt` attached, hopefully this gives you enough info

[linux-6.14-journalctl-daily-fw.25.02-2025-03-27.txt](https://github.com/user-attachments/files/19493911/linux-6.14-journalctl-daily-fw.25.02-2025-03-27.txt)

Sean-StarLabs, 2025-03-27:
Thanks - one thing we can try, will try to do early next week

BlackIkeEagle, 2025-03-28:
bisected to the first problematic commit:

```
git bisect start '--' 'drivers/net/wireless/intel/iwlwifi'
# status: waiting for both good and bad commits
# good: [ffd294d346d185b70e28b1a28abe367bbfe53c04] Linux 6.13
git bisect good ffd294d346d185b70e28b1a28abe367bbfe53c04
# status: waiting for bad commit, 1 good commit known
# bad: [2014c95afecee3e76ca4a56956a936e23283f05b] Linux 6.14-rc1
git bisect bad 2014c95afecee3e76ca4a56956a936e23283f05b
# bad: [1dde1f316d476e863cbd74ccecd6301fe6d9b37e] wifi: iwlwifi: move fw_ver debugfs to firmware runtime
git bisect bad 1dde1f316d476e863cbd74ccecd6301fe6d9b37e
# good: [9b45ba3976945e8d53f2dd40541a66c690f12286] wifi: iwlwifi: pcie: Add support for new device ids
git bisect good 9b45ba3976945e8d53f2dd40541a66c690f12286
# good: [8c4c0485de3453c2dc27a0e1c84890afee1c4527] wifi: iwlwifi: context-info: add kernel-doc markers
git bisect good 8c4c0485de3453c2dc27a0e1c84890afee1c4527
# good: [99baaf924a30b240622a42fdab0fff3ffa8138cc] wifi: iwlwifi: mvm: clean up FW restart a bit
git bisect good 99baaf924a30b240622a42fdab0fff3ffa8138cc
# bad: [7391b2a4f7dbb7be7dd763bc87506c10f570a8d3] wifi: iwlwifi: rework firmware error handling
git bisect bad 7391b2a4f7dbb7be7dd763bc87506c10f570a8d3
# good: [2d15d213e3d83108f1e29a80b230f9adb8b6e0c9] wifi: iwlwifi: mvm: restart device through NMI
git bisect good 2d15d213e3d83108f1e29a80b230f9adb8b6e0c9
# first bad commit: [7391b2a4f7dbb7be7dd763bc87506c10f570a8d3] wifi: iwlwifi: rework firmware error handling
```

```
7391b2a4f7dbb7be7dd763bc87506c10f570a8d3 is the first bad commit
commit 7391b2a4f7dbb7be7dd763bc87506c10f570a8d3
Author: Johannes Berg <johannes.berg@intel.com>
Date:   Fri Dec 27 10:01:11 2024 +0200

    wifi: iwlwifi: rework firmware error handling
    
    In order to later add the ability to do deeper resets of the
    device when it crashes, first restructure the firmware error
    handling. Instead of having just a single nic_error() method
    that handles all, split it:
     - nic_error() just handles and prints the error itself,
     - dump_error() synchronously creates an error dump, and
     - sw_reset() will be called to request doing a SW reset.
    
    This changes the architecture so that the transport is now
    responsible for deciding how to do the reset, and therefore
    the handling of reprobe if error occurs during reconfig
    moves there, which necessitates adding a method there that
    notifies the transport that the recovery was completed.
    
    Actually introducing the model under which deeper resets can
    be done will be in future patches.
    
    Signed-off-by: Johannes Berg <johannes.berg@intel.com>
    Signed-off-by: Miri Korenblit <miriam.rachel.korenblit@intel.com>
    Link: https://patch.msgid.link/20241227095718.6d4f741ae907.I96a9243e7877808ed6d1bff6967c15d6c24882f0@changeid
    Signed-off-by: Johannes Berg <johannes.berg@intel.com>

 drivers/net/wireless/intel/iwlwifi/dvm/mac80211.c    |  11 ++++++++++
 drivers/net/wireless/intel/iwlwifi/dvm/main.c        |  27 ++++++++++++++++-------
 drivers/net/wireless/intel/iwlwifi/iwl-op-mode.h     |  53 ++++++++++++++++++++++++++++++++++++++++++--
 drivers/net/wireless/intel/iwlwifi/iwl-trans.c       | 109 ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++--
 drivers/net/wireless/intel/iwlwifi/iwl-trans.h       |  56 ++++++++++++++++++++++++++++++++++++++++++++++-
 drivers/net/wireless/intel/iwlwifi/mvm/mac80211.c    |   1 +
 drivers/net/wireless/intel/iwlwifi/mvm/ops.c         | 143 ++++++++++++++++++++++++++++++++++++++++++++++++-----------------------------------------------------------------------
 drivers/net/wireless/intel/iwlwifi/pcie/trans-gen2.c |  11 ++++++++--
 drivers/net/wireless/intel/iwlwifi/pcie/tx-gen2.c    |   1 +
 drivers/net/wireless/intel/iwlwifi/pcie/tx.c         |   1 +
 10 files changed, 312 insertions(+), 101 deletions(-)
```

Will try later if a revert is possible or not, but this is a fairly big change so probably not straight forward

BlackIkeEagle, 2025-03-29:
Ok have to say I'm not qualified to figure this out any further it seems :)

A simple revert is not possible

I also attach a journal of the 6.12 kernel so it might give some clue compared to the 6.14 boot

[linux-6.12.journalctl.txt](https://github.com/user-attachments/files/19521900/linux-6.12.journalctl.txt)

sim31, 2025-04-03:
Does this affect other models? Care about Starbook mk VI specifically.

Sean-StarLabs, 2025-04-03:
No, only CNVi

macmpi, 2025-04-06:
Having similar failure since 2 days with MkIV (25.02) on 5.15.0-136-generic #147~20.04.1-Ubuntu.
```
[  317.734776] iwlwifi 0000:00:14.3: Microcode SW error detected. Restarting 0x0.
[  317.734881] iwlwifi 0000:00:14.3: Start IWL Error Log Dump:
[  317.734886] iwlwifi 0000:00:14.3: Transport status: 0x0000004A, valid: 6
[  317.734893] iwlwifi 0000:00:14.3: Loaded firmware version: 66.f1c864e0.0 QuZ-a0-hr-b0-66.ucode

```
Enclosing [iwlwifi](https://github.com/user-attachments/files/19621485/iwlwifi.txt) entries from dmesg

macmpi, 2025-04-07:
>  so it'll need to be fixed linux side.

In my case the issue materialized a few days back only , and after a restart which did not actually changed kernel version (some other packages had been installed, mostly relative hardware headers & tzdata in subsequent updates): wifi worked with same 5.15.0-136 kernel prior to that (used it several days before that deadly restart).

I assume Starlabs machines may not the only ones affected (i.e. same wifi chipset), and it spawns a bunch of kernel releases already.
Have you seen any relevant reports elsewhere, and in particular fix being worked-out on the linux side?
Anything that can be tracked?

Sean-StarLabs, 2025-04-07:
@macmpi This is not the same issue - I think from the snippet, yours is https://forums.linuxmint.com/viewtopic.php?t=295219

macmpi, 2025-04-07:
@Sean-StarLabs  thanks for feedback.
Tried suggested method with no luck: it still crashes right after association.
Also tried removing` /etc/modprobe.d/intel-microcode-blacklist.conf`  `iwlwifi.conf` `starlabs.conf`, and add latest `iwlwifi-QuZ-a0-hr-b0-77.ucode` from https://gitlab.com/kernel-firmware/linux-firmware.
Baseline installed firmware files are:
```
ls -la /lib/firmware/ | grep QuZ-a0-hr-b0-
-rw-r--r--   1 root root 1105668 avril 21  2023 iwlwifi-QuZ-a0-hr-b0-48.ucode
-rw-r--r--   1 root root 1103952 avril 21  2023 iwlwifi-QuZ-a0-hr-b0-50.ucode
-rw-r--r--   1 root root 1223140 avril 21  2023 iwlwifi-QuZ-a0-hr-b0-55.ucode
-rw-r--r--   1 root root 1299660 avril 21  2023 iwlwifi-QuZ-a0-hr-b0-59.ucode
-rw-r--r--   1 root root 1334804 avril 21  2023 iwlwifi-QuZ-a0-hr-b0-63.ucode
-rw-r--r--   1 root root 1344216 avril 21  2023 iwlwifi-QuZ-a0-hr-b0-66.ucode
-rw-r--r--   1 root root 1347524 avril 21  2023 iwlwifi-QuZ-a0-hr-b0-67.ucode
-rw-r--r--   1 root root 1355800 avril 21  2023 iwlwifi-QuZ-a0-hr-b0-68.ucode
-rw-r--r--   1 root root 1364308 avril 21  2023 iwlwifi-QuZ-a0-hr-b0-71.ucode
```

Sean-StarLabs, 2025-04-07:
@macmpi  Feel free to ping support

BlackIkeEagle, 2025-04-12:
Are there any updates related to this? firmware / linux kernel patch?

AndrzejTunkiel, 2025-04-14:
> Are there any updates related to this? firmware / linux kernel patch?

It seems that this is patched and being upstreamed - https://bugzilla.kernel.org/show_bug.cgi?id=219968

Most relevant:

> Johannes Berg 2025-04-10 20:36:52 UTC
> great thanks, we'll send the patch on the way the usual process

> might take a bit since various spring vacations are upcoming

BlackIkeEagle, 2025-04-14:
I can confirm that with the Daily firmware and this: https://web.git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/commit/?id=5f05c14e7c198415abe936514a6905f8b545b63b patch the wifi on my Starlite MK V works with linux 6.14

BlackIkeEagle, 2025-04-14:
This FW reset timeout is the coreboot part which is not optimal? Since this also appears on older kernels?

```
iwlwifi 0000:00:14.3: timeout waiting for FW reset ACK (inta_hw=0x0)
```

BlackIkeEagle, 2025-04-15:
> So just for confirmation; this is fixed, and I can safely upgrade my kernel on archlinux?

No, you need to rebuild with the patch from the linux-next tree referenced earlier. For now it is best to stick to LTS at this moment

Sean-StarLabs, 2025-04-16:
> This FW reset timeout is the coreboot part which is not optimal? Since this also appears on older kernels?
> 
> ```
> iwlwifi 0000:00:14.3: timeout waiting for FW reset ACK (inta_hw=0x0)
> ```

Nope - that's Linux, one of those bugs that hasn't seemed to go away but doesn't cause an major issues

stalkerg, 2025-04-17:
Hmm, in my case, WiFi is working, but the Bluetooth device does not exist... I even tried to use 6.13, but the problem still remains. Can it be related?

meeuw, 2025-04-19:
FYI: Fedora 42 has been released with `Linux version 6.14.2-300.fc42.x86_64`

Wifi doesn't work with the Live ISO: https://fedoraproject.org/workstation/download

nicolaslpf, 2025-04-19:
Same problem here with ubuntu after upgrading to 25.04 with kernel 6.14.0-15.15, I had to go back to kernel 6.11.0-24.24. Hope Ubuntu releases kernel 15 as soon as stable is released.

meeuw, 2025-04-20:
I cowardly switched back to [kernel-6.13.11-200.fc41](https://bodhi.fedoraproject.org/updates/FEDORA-2025-2587f342c2) (using it with Fedora 42)

I hope 6.15 will contain the patch, might want to try a release candidate...

nicolaslpf, 2025-04-21:
6.15-rc3 is out

BlackIkeEagle, 2025-04-25:
Fixed in 6.14.4

nicolaslpf, 2025-04-25:
Yeah confirmed here on ubuntu. Solved in 6.14.4

czinehuba, 2025-04-27:
Yes. Confirmed .works with Arch 6.14.4

closed 2025-04-27 by BlackIkeEagle

nicolaslpf, 2025-05-31:
> > Yeah confirmed here on ubuntu. Solved in 6.14.4
> 
> Do you also mean it got solved it Ubuntu? As of Ubuntu 25.04, fully update (through ethernet via a docking station), I still have this problem. Did you manually compile a newer version of the kernel?

Nope no needs... just instal mainline kernels package and you select the kernel. Is not an official ubuntu kernel but it works great. Then you boot with this kernel.