StarLabsLtd/firmware #270: [StarBook MkVII-U] NPU Disabled in firmware

makinikm, 2025-03-29:
NPU (vpu) is disabled in coreboot 25.02 on StarBook MkVII-U B7-U (v1.0) - according to [this...](https://github.com/StarLabsLtd/coreboot/blob/4e06bdf0952934101566c8be0b7ffb5d9f14e1d1/src/soc/intel/meteorlake/chipset.cb#L67C3-L67C9)

---

Steps to reproduce:
1. Attempt to setup NPU driver as per [upstream...](https://github.com/intel/linux-npu-driver/releases)
2. Module loaded (intel_vpu), but /dev/accel* device(s) not present after reboot, nor listed by lspci...
3. No vpu-related dmesg output even with debug log module option (ivpu_log_level=1) 

---

Expected Behavior:
Either to have a menu option in coreboot for enabling the NPU, or enabled by default on supporting models

---

System Information:
- **OS: Linux Mint 22.1 Xia base: Ubuntu 24.04 noble
- **Kernel: 6.11.0-21-generic arch: x86_64
- **Product SKU: B7-U
- **BIOS Version: 25.02
- **EC Release: 25.2

--- 

Additional Context:
Not really relevant, but the NPU issue was revised by system76 guys [here](https://github.com/system76/firmware-open/issues/580) with the decision to enable it for the branch eventually... 
Perhaps a disabled menu option could be a solution if not desired on by default, for some reason. Otherwise - why disable a part of the CPU that accelerates running local LLMs? ;)

---

makinikm, 2025-03-31:
Hi, not sure if to open another issue about it or could fit here... 
The GNA (Gaussian & Neural Accelerator) is also disabled in similar fashion right next to the vpu 25.02 [here...](https://github.com/StarLabsLtd/coreboot/blob/4e06bdf0952934101566c8be0b7ffb5d9f14e1d1/src/soc/intel/meteorlake/chipset.cb#L65)
Its used via same upstream toolkit for [NN audio acceleration](https://github.com/intel/gna). Was there a security concern perhaps in disabling it as with ME? Doesn't seem to be a part of ME by design though. Maybe a power consumption/heating?
Could you also consider adding an option or enabling it for mlk?

Sean-StarLabs, 2025-04-01:
https://review.coreboot.org/c/coreboot/+/87055

> Maybe a power consumption/heating?

Just illogical to switch everything on by default

makinikm, 2025-04-01:
Thank You :pray: