StarLabsLtd/firmware #271: StarLite V Battery not charging

rogercreagh, 2025-03-29:
Suddenly today the battery has stopped charging. I turned the machine off about 6hrs ago with the battery at around 50%. I plugged in the charger and the red light lit up as normal.

When I got back to it the light was blue with a hint of red (as it usually is when near fully charged) but when I unplugged it and turned it on it was shutting down after about 5 seconds of the login prompt appearing.

Plugged the charger in and it came back to life but showing battery level 0%. The charge light is now blue (ie it thinks its fully charged) but the gauge still says 0% and if I unplug the charger it shuts down.

Is the battery actually dead and not charging, or has something gone wrong in the firmware? The settings power battery level is showing it jumping around from 0% not charging to various random levels at intervals of a few seconds.

The utilities power statistics says 0.0W  with the state jumping around. The charge graph shows it at 50% 6.5hrs ago and nothing since.

Ubuntu 24.04.2  Firmware 24.10 Kernel Linux 6.11.0-21-generic.

I see elsewhere suggestions that the calibration might need a reset - how do I do this? Is the a conf file somewhere that needs deleting?

Sean-StarLabs, 2025-03-29:
That's fixed in 25.02 but if you shutdown, disconnect the charger, wait 12 seconds until the LEDs flicker, reconnect the charger until you get a solid charging LED, it'll work

rogercreagh, 2025-03-30:
Thanks @Sean-StarLabs that worked. Can't update firmware till bluetooth is fixed so I'll try and remember this if it happens again.

closed 2025-03-30 by rogercreagh