StarLabsLtd/firmware #274: [StarBook Mk VI - AMD] Support for changing charge thresholds via sysfs interface

alternateved, 2025-04-06:
Hello!

Would it be possible to enable support for changing charge thresholds via sysf interface so that latest Gnome feature would work? 

Feature is explained here: 
https://vdwaa.nl/gnome-upower-charge-thresholds.html

Relevant MR is here:
https://gitlab.freedesktop.org/upower/upower/-/merge_requests/208

Thanks for your help!

NOTE: I wasn't sure if I should create that query/feature request via discussions or issues so in the end I've decided to do this here.

Sean-StarLabs, 2025-04-06:
The duplicate of #108, a short version, creates a dependency, so I'm sticking with it being configurable in edk2.

closed 2025-04-06 by Sean-StarLabs

alternateved, 2025-04-06:
I see, thanks! That is what I wanted to know.