StarLabsLtd/firmware #280: [StarBook MkVIr2-Intel] After update to 25.02 it's freez on StarLab logo

stalkerg, 2025-04-17:
### **Describe the bug**
I did update bios by this ROM:
https://github.com/StarLabsLtd/firmware/tree/main/StarBook/MkVIr2-Intel/coreboot/25.02

and it seems like done correct (first power on I saw led blinks and etc). 
Unfortunately during boot it's freeze. I can see StarLab logo plus text (about F2 and bios menu) for a few seconds and after only StarLab logo and nothing more. 

Seems like it's kind of brick? 
Thanks

stalkerg, 2025-04-17:
> With flashrom?

yes. by `flashrom -p internal -w /tmp/coreboot.rom -i bios --ifd -n -N`

stalkerg, 2025-04-17:
It seems like I should buy CH341A...

Sean-StarLabs, 2025-04-17:
> It seems like I should buy CH341A...

Maybe - only thing you can really try is:
1. Disconnect the battery
2. Remove the SSD
3. Wait a couple of minutes
4. Connect the charger - see what happens

If that doesn't work, then yes, programmer required

stalkerg, 2025-04-17:
Okey! I got EFI shell.

stalkerg, 2025-04-17:
and after all bios is working!!!

stalkerg, 2025-04-17:
Seems like it's freezes if I put SSD. First time I saw some EFI mapping info and after it's became freeze.
I can run bios menu, but inside if I choose "boot manager" it's freezes.

closed 2025-04-17 by Sean-StarLabs

stalkerg, 2025-04-17:
@Sean-StarLabs some update, maybe it's still bug in firmware:
1. Seems like SSD itself totally fine. I used NVMe to USB on my main PC and check GPT partition plus fsck with badblocks. 
2. gdisk show only a few warnings:
```
Warning: There is a gap between the main partition table (ending sector 33)
and the first usable sector (2048). This is helpful in some exotic configurations,
but is unusual. The util-linux fdisk program often creates disks like this.
Using 'j' on the experts' menu can adjust this gap.

Caution: Partition 2 doesn't end on a 2048-sector boundary. This may
result in problems with some disk encryption tools.

No problems found. 7118 free sectors (3.5 MiB) available in 2
segments, the largest of which is 5070 (2.5 MiB) in size.
```
3. My table is next now:
```
label: gpt
label-id: 38955405-B890-40D7-92D5-24476D49A7F6
device: /dev/sdc
unit: sectors
first-lba: 2048
last-lba: 1953525134
sector-size: 512

/dev/sdc1 : start=        4096, size=      614400, type=C12A7328-F81F-11D2-BA4B-00A0C93EC93B, uuid=3FD7081E-D6F9-4788-BC2A-43D1B187FD99
/dev/sdc2 : start=      618496, size=  1952901569, type=0FC63DAF-8483-4772-8E79-3D69D8477DE4, uuid=F1C184A8-989F-4210-B41B-DF3A692E7EBC, name="root"
```
4. I can run linux from liveusb without issue. 
5. On screen I can see next:

[image]

Question: Is it possible because of grub/efi file are broken? Should I try to install it again? I can try boot from SSD by qemu or by USB. Will see...

stalkerg, 2025-04-17:
New update! I can boot to my system thru USB adapter, just NVMe slot is not working correctly.
Seems like it's firmware issue.

stalkerg, 2025-04-17:
I put a different SSD to NVMe and it's stuck even more worse. This SSD didn't contain any boot things, just data partition.

stalkerg, 2025-04-17:
Okey, I fleshed 24.10 and it's working again.