StarLabsLtd/firmware #281: [StarBook MkVIr2-Intel] NVMe doesn't work under 25.02

stalkerg, 2025-04-19:
### **Describe the bug**
With 25.02 firmware if you have any (even empty) ssd in NVMe slot the bios will be freeze during boot. You have chance open menu but if you try enter to "Boot Manager" it freeze again. 

---

### **Steps to Reproduce**
Detailed steps to reproduce the issue:
1. Flash 25.02
2. Reboot

Sean-StarLabs, 2025-04-19:
Please don't log bugs against unstable builds

closed 2025-04-19 by Sean-StarLabs

stalkerg, 2025-04-19:
Sorry, how can I check if it is stable or not? I am confused. 
Also, I think even if it's unstable the bug report should be valuable, no? 
Anyway, I solved my issues, if you are not interested in my help it's up to you. 

Regards,

Sean-StarLabs, 2025-04-21:
> Sorry, how can I check if it is stable or not? I am confused.

https://fwupd.org/lvfs/devices/com.starlabs.B62-I.coreboot

> Also, I think even if it's unstable the bug report should be valuable, no?

LVFS only serves stable and testing, anything outside of that isn't really looked at - unless it's a specific bug we're sending versions out for