StarLabsLtd/firmware #282: Battery not charging after installing 25.03 (or 25.02 ?)

lducas, 2025-04-22:
### **Describe the bug**
Since trying installing firmware 25.03 following instructions found here (https://github.com/StarLabsLtd/firmware/issues/267), my starbook Mk VII does not appear to recharge battery anymore. Given the data below I'm unsure wether it was 25.02 or 25.03 that was installed.

### **System Information**
Provide your system details:
- **OS:**  Ubuntu 24.04.2 LTS.
- **Kernel Version:** 6.11.0-21-generic
- **Product SKU:** B7-N
- **BIOS Version:** 25.02
- **EC Release:** 25.2

---

### **Logs**
Include any relevant logs that might help diagnose the issue:
- **Dmesg Output:** Attach or paste the relevant portion of `dmesg` logs.
- **Other Logs:** Specify and include logs from related subsystems if applicable.

---

### **Additional Context**
 I was trying to get HDMI screen to work, and now my whole laptop has an hour left to live :(

lducas, 2025-04-22:
Re-executing the whole procedure seems to have solved the matter.

lducas, 2025-04-22:
Charging failed again after I tried plugging a HDMI screen via a USB-C adapter.
And re-flashing the rom fixed it once again.

Sean-StarLabs, 2025-04-22:
Issue is only in the daily build?

lducas, 2025-04-22:
The issue is appearing again: power is plugged in, ubuntu says it is charging, and yet charging percentage is slowly decreasing. I've increased charging speed from 0.5C to 1.0C in the bios, and the issue remain.

I indeed download the daily_build via:
`wget https://raw.githubusercontent.com/StarLabsLtd/firmware/refs/heads/main/daily_builds/$(cat /sys/class/dmi/id/product_sku).rom -O /tmp/coreboot.rom`

I'm happy to try a stable build, but I have yet to find the procedure for doing so.

Sean-StarLabs, 2025-04-22:
Did it happen in the stable build (the one it was sent with)?
What's the charge level?

lducas, 2025-04-22:
No I hadn't seen that issue before firmware upgrade. 
Current charge level is 41% ... Oh, 40% now ...
I'll try to let it charge whith power off to see if I'm not just bricking my machine as we speak.

Sean-StarLabs, 2025-04-22:
Same in both ports?

lducas, 2025-04-22:
What do you mean both ports ? I use the one on the top left, with a "power plug" logo. 
Is there another one that can be used for charging ?
The charging led is on and red.

lducas, 2025-04-22:
Could you point me to instructions on how to revert to a stable build ?

Sean-StarLabs, 2025-04-22:
> What do you mean both ports ?

There are two Type-C ports

waylon531, 2025-04-22:
I have the same issue with a nightly build and the MkVII, it is fixed if I charge off the secondary USBC port. Swapping chargers doesn't affect it at all. Is there any good way to pull logs for this kinda issue?

lducas, 2025-04-22:
Indeed, the other port seems to be charging fine, I wouldn't have dared pushing power through there, given that there is no power plug logo. I guess I should learn about Type-C specs. 

This remains a temporary solution, and I would appreciate instructions on how to revert to a stable version (maybe I should have thought of that before downloading a daily build).

Sean-StarLabs, 2025-04-23:
Fixed with a13c043a09526b7b7b54e5cf9ba1a9a7c02fe993

closed 2025-04-23 by Sean-StarLabs

Sean-StarLabs, 2025-05-02:
> The daily builds are for the exact same thing as the cab, right?

No. 25.04 hasn't even been tested; without knowing if it works or not, hard to say what the problem is.