StarLabsLtd/firmware #283: Can't upgrade UEFI dbx from 471 to 20241101 on lite mk3

soc7, 2025-04-22:
### **Describe the bug**
Provide a clear and concise description of the issue, including any error messages, unexpected behaviors, or system crashes.
 I see un upgrade available of UEFI dbx but after running the upgrade on fwupdmgr and rebooting it still has the old version and offers the same upgrade on fwupd.

### **Steps to Reproduce**
Detailed steps to reproduce the issue:
1. Go to terminal
2. Perform fwupdmgr update
3. Observe upgrade succesful then reboot

Be as specific as possible to help us replicate the issue.

---

### **Expected Behavior**
Describe what you expected to happen instead of the observed behavior.

---

### **Screenshots**
If applicable, upload screenshots or screen recordings to help visualize the issue.

---

### **System Information**
Provide your system details:
- **OS: debian 12
- **Kernel Version:6.1.33
- **Product SKU: I3
- **BIOS Version: 2.1.0
- **EC Release: 1.14

---

### **Logs**
$ fwupdmgr update
Dispositivi senza aggiornamenti firmware:
 • Star Drive SATA SSD
Dispositivi con l'ultima versione disponibile del firmware:
 • System Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Aggiornare UEFI dbx da 471 a 20241101?                                       ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This updates the list of forbidden signatures (the "dbx") to the latest      ║
║ release from Microsoft.                                                      ║
║                                                                              ║
║ An insecure version of Howyar's SysReturn software was added, due to a       ║
║ security vulnerability that allowed an attacker to bypass UEFI Secure Boot.  ║
║                                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Eseguire l'operazione? [Y|n]:

### **Additional Context**
Provide any other information or context that might help in debugging this issue. This could include:
- Previous steps taken to resolve the issue.
- Workarounds attempted, if any.
- Frequency of the issue (e.g., "Always," "Intermittent").

---

Sean-StarLabs, 2025-04-22:
Duplicate of #266

closed 2025-04-22 by Sean-StarLabs

Sean-StarLabs, 2025-04-22:
Sorry, missed that. coreboot, new version yes, but it'll still be classed as beta for s3 issues.

soc7, 2025-04-22:
No problem thanks for the info. Do you think someday there will be a stable version of coreboot for the lite mk3?