StarLabsLtd/firmware #288: [StarLite Mk V] Feature request: Expose BIOS Settings via UEFI Variables for Runtime Access (efivarfs)

rbm78bln, 2025-04-30:
Hi @Sean-StarLabs,
Hi StarLabs team,

I'd like to request firmware support for accessing key BIOS settings via UEFI variables with `EFI_VARIABLE_RUNTIME_ACCESS`, making them accessible through `efivarfs` under Linux. This would enable both reading and writing these specific settings directly from userspace, enabling automatic/scripted power- and performance-management — especially improving user experience on Linux-first devices like the StarLite Mk V.

### Proposed settings to expose:
- **Power Profile** (`Powersaving`, `Balanced`, `Performance`)
- **Maximum Battery Charge Level**
- **Battery Charging Speed**
- **Memory Speed**
- **Power LED Brightness** (`Off`, `Dim`, `Regular`)

#### Proposed UEFI variable flags:
- `EFI_VARIABLE_NON_VOLATILE`
- `EFI_VARIABLE_BOOTSERVICE_ACCESS`
- `EFI_VARIABLE_RUNTIME_ACCESS`

Immediate application of settings would be ideal where possible, but delayed effect (e.g. after reboot) is acceptable for settings with firmware or hardware-level constraints such as memory timings.

This addition would significantly enhance system manageability on Linux and reduce reliance on manual interaction with the BIOS UI. It also fits well with your target audience of Linux power users and developers.

Kind regards,  
Ralf

Anachron, 2025-10-14:
I'd also LOVE to be able to set the battery charge start and stop values.

Sean-StarLabs, 2025-10-24:
Exporting?

Anachron, 2025-10-25:
> > I'd also LOVE to be able to set the battery charge start and stop values.
> 
> That's already an option

In Linux? Not here.