StarLabsLtd/firmware #290: Starlite V Firmware 25.04 Testing Feedbacl

thebatdan, 2025-05-03:
### **Describe the bug**
I noticed that a testing version of fw 25.04 was released on LVFS. I decided to give it a go given it indicates that it fixes the inconsistent charger detection (as reported #265 and #197), which is still an issue with my device.

Unfortunately, the inconsistent charger detection still occurs. Additionally, only the upper usb-c port now works for charging, as described in #282. Downgrading to 25.02 restores charging on the lower port. 

---

### **Steps to Reproduce**
Detailed steps to reproduce the issue:
1. Enable fwudmgr testing repo as per https://support.starlabs.systems/kb/firmware/getting-started
2. Install firmware 25.04, making sure both bios and firmware versions are updated
3. Attach the charger to the lower usb-c port
4. After no charge, attach to upper usb-c port

Be as specific as possible to help us replicate the issue.

---

### **Expected Behavior**
Describe what you expected to happen instead of the observed behavior.
* Both usb-c ports support charing
* Charging completes uninterrupted
---

### **System Information**
Provide your system details:
- **OS:** Fedora 42
- **Kernel Version:** 6.14.4
- **Product SKU:** I5
- **BIOS Version:** 25.04
- **EC Release:** 25.4

---

rbm78bln, 2025-05-10:
Yup, can confirm.

jothgard, 2025-05-15:
did update to 24.05 stable, my device cannot see any battery at all, did downgrade to 24.04, and there my battery was :)... So need  battery fix for 24.05.

Sean-StarLabs, 2025-05-15:
> did update to 24.05 stable

24.05 is not stable

> my device cannot see any battery at all

You flashed the wrong firmware

jothgard, 2025-05-16:
> > did update to 24.05 stable
> 
> 24.05 is not stable
> 
> > my device cannot see any battery at all
> 
> You flashed the wrong firmware

Do men "stable" when i saw your official release update for 25.05.

Oh i saw now that I did type 24.05, i do mean 25.05 :)

Sean-StarLabs, 2025-05-16:
Ah yeah, we're both talking about 25.05