StarLabsLtd/firmware #295: [LabTop MK IV] Wont charge after 25.05, noisy fan

macmpi, 2025-05-15:
Since 25.05 firmware update, L4 does not charge anymore and fan is blowing even in low activity.
Charger is detected in status bar.
Left several hours, charge stayed at about 30%.

### **System Information**
Provide your system details:
- **OS: Ubuntu 22.04 (elementary 8.0.1)**
- **Kernel Version: 6.14.6**
- **Product SKU: L4**
- **BIOS Version: 25.05**
- **EC Release: 25.5**

Sean-StarLabs, 2025-05-15:
>  L4 does not charge

Need to know:
* What charger
* What cable
* If this is the first run after updating (i.e. has it reached G3 since being on 25.05)

>  fan is blowing even in low activity

Need to know:
* temperature (`sensors`)
* which fan curve

macmpi, 2025-05-15:
>     * What charger
> 
>     * What cable

No change from 5mn before when in 25.02 (or previous versions): 100W and 100W capable cable.

>     * If this is the first run after updating (i.e. has it reached G3 since being on 25.05)

Made several _power-off then wait G3_ tests.

>     * temperature (`sensors`)

Fan went crazy (loud) at first boot after update.
After reboots, it keeps blowing constantly even at low level(unless suspended)
      
```
$ sensors
nvme-pci-0100
Adapter: PCI adapter
Composite:    +34.9°C  (low  =  -0.1°C, high = +69.8°C)
                       (crit = +89.8°C)

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +35.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +35.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +35.0°C  (high = +100.0°C, crit = +100.0°C)

pch_cannonlake-virtual-0
Adapter: Virtual device
temp1:        +33.0°C  

BAT0-acpi-0
Adapter: ACPI interface
in0:           7.46 V  
curr1:         0.00 A  
```

>     * which fan curve

dunno, untouched.

Sean-StarLabs, 2025-05-15:
> No change from 5mn before when in 25.02 (or previous versions): 100W and 100W capable cable.

Sorry, I have no idea what you mean by this. Can you just let me know the make/model of the charger/cable?

> Fan went crazy (loud) at first boot after update.

It should

Sean-StarLabs, 2025-05-15:
@MrChromebox FYI, caused by ac9a49f226e94c7b8adec7886d216ff23ff4692d

macmpi, 2025-05-15:
> Sorry, I have no idea what you mean by this. Can you just let me know the make/model of the charger/cable?

 [Baseus 100W GaN2](https://www.amazon.fr/dp/B09HXRSX21) bought as a replacement to original PSU and cable which failed/fried.
Very efficient with L4 (until 25.05 update), and many other products.

Does PECI relate to charging too?

Sean-StarLabs, 2025-05-15:
> Does PECI relate to charging too?

No

macmpi, 2025-05-15:
Charging is key issue here after update: why removing from ticket topic?

Sean-StarLabs, 2025-05-15:
Because I can replicate that and it's clear what the problem is. 

Charging needs investigating.

macmpi, 2025-05-15:
Now L4 is below 30%, how can I downgrade firmware before I'm out of juice?

Sean-StarLabs, 2025-05-15:
```
wget https://github.com/StarLabsLtd/firmware/raw/refs/heads/main/LabTop/MkIV/coreboot/24.10/24.10.rom -O /temp/coreboot.rom
sudo flashrom -p internal -w /temp/coreboot.rom -i EC --fmap -n -N
```

macmpi, 2025-05-15:
Thanks for prompt answer. where can I get proper `flashrom` tool?

Sean-StarLabs, 2025-05-15:
What distro? Most have it in their repos

macmpi, 2025-05-15:
ok, thought it might have some specific variant. `sudo apt install flashrom` ok

macmpi, 2025-05-15:
after this, I'm at:
```
$ cat /sys/class/dmi/id/ec_firmware_release
24.10
$ cat /sys/class/dmi/id/bios_version
25.05
```
expected?

MrChromebox, 2025-05-15:
> [@MrChromebox](https://github.com/MrChromebox) FYI, caused by ac9a49f226e94c7b8adec7886d216ff23ff4692d

26997e8a94456f2182e68a466680f0584b0ba281 should fix, but don't have a PECI-capable device here to test

macmpi, 2025-05-16:
> Charging needs investigating.

@Sean-StarLabs  let me know if I can provide any useful information on charging part.
Can confirm 24.10 EC with 25.05 firmware (resulting from your [tip](https://github.com/StarLabsLtd/firmware/issues/295#issuecomment-2885134452)) is charging fine when system is on (improvement).
However led still does not turn red if connecting to charger when G3

Also curious about `24.10 EC` and `25.05 EC` difference on `iwlwifi` stances (latter has none)...

MrChromebox, 2025-05-16:
> Also curious about `24.10 EC` and `25.5 EC` difference on `iwlwifi` stances (latter has none)...

the EC has nothing to do with the WiFi module temp reporting, that's a function of the iwlwifi driver IIRC

macmpi, 2025-05-16:
`sensors` command reports are what they are: only difference between the 2 reports is different EC (same system, same kernel & drivers,...)

MrChromebox, 2025-05-16:
> `sensors` command reports are what they are: only difference between the 2 reports is different EC (same system, same kernel & drivers,...)

`sensors` reports the iwlwifi virtual device regardless of EC version used. You can easily check this by re-flashing the 25.05 EC (or flashing the one below with the PECI temp fixed)

[l4_ec_25.05_peci_fix.zip](https://github.com/user-attachments/files/20258975/l4_ec_25.05_peci_fix.zip)

`sudo flashrom -p internal -w --fmap -i EC:ec.bin -N`

then allow the device to go into G3

macmpi, 2025-05-16:
```
$ sudo flashrom -p internal -w --fmap -i EC:ec.bin -N
flashrom 1.4.0-devel on Linux 6.14.6-061406-generic (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Warning: Supplied image file name starts with -
Region requested (with -i "EC"), but no layout data is available.
```

MrChromebox, 2025-05-16:
well that's annoying, must be an older version of flashrom since that's what I use here to update

macmpi, 2025-05-20:
Any suggested workaround to flash fixed 25.5 EC with shipping `flashrom` tool?
Would that work better if bundled with coreboot for re-installing the whole lot?

MrChromebox, 2025-05-20:
here's a coreboot 25.05 image with the fixed EC integrated

[coreboot_25.05_l4_20250520.zip](https://github.com/user-attachments/files/20348469/coreboot_25.05_l4_20250520.zip)

download, unzip, then flash:

`sudo flashrom -p internal -w coreboot_25.05_l4_20250520 --fmap -i EC -N`

shutdown / go into G3

macmpi, 2025-05-20:
Thanks, worked fine.
However charging issue is still there with that EC (led ON, charging icon, but not charging: `curr1:  0.00 A` below ), while it charged with previous 24.10 EC

```
$ cat /sys/class/dmi/id/ec_firmware_release
25.5

$ cat /sys/class/dmi/id/bios_version
25.05

$ sensors
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +47.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +47.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +44.0°C  (high = +100.0°C, crit = +100.0°C)

nvme-pci-0100
Adapter: PCI adapter
Composite:    +40.9°C  (low  =  -0.1°C, high = +69.8°C)
                       (crit = +89.8°C)

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:            N/A  

pch_cannonlake-virtual-0
Adapter: Virtual device
temp1:        +37.0°C  

BAT0-acpi-0
Adapter: ACPI interface
in0:           7.81 V  
curr1:         0.00 A  
```

thorsten-occup, 2025-05-30:
I have the same issue with charging the LapTop Mk IV. I look forward to a fix and thank you for your hard work!

Sean-StarLabs, 2025-05-30:
> I have the same issue with charging the LapTop Mk IV. 

Also a third party charger?

thorsten-occup, 2025-05-30:
With the StarLabs charger, that was included in the box and any other charger.

thorsten-occup, 2025-06-01:
> > I have the same issue with charging the LapTop Mk IV. 
> 
> Also a third party charger?

If you need any diagnostic data, just drop me a line.

Sean-StarLabs, 2025-06-13:
Fixed with 4af8cd6f857f4018b0d5e270f6724146d3709f96

closed 2025-06-13 by Sean-StarLabs