StarLabsLtd/firmware #296: Microphone with noise

soc7, 2025-05-19:
### **Describe the bug**
Provide a clear and concise description of the issue, including any error messages, unexpected behaviors, or system crashes.

While doing a call or video call the person on the other side complains that there are some strong noise, I tested jitsi, google and microsoft video-calls, all with same result, actually it happens since a long time on this PC. With headphones connected to audiojack or internal microphone, same result

### **Steps to Reproduce**
Detailed steps to reproduce the issue:
1. Go to any videocall service
2. Perform a videocall
3. Observe noise on the other side, caused by the L3 microphone

Be as specific as possible to help us replicate the issue.

---

### **Expected Behavior**
Describe what you expected to happen instead of the observed behavior.

### **Screenshots**
If applicable, upload screenshots or screen recordings to help visualize the issue.

---

### **System Information**
Provide your system details:
- **OS:** Arch Linux but it did the same behavior on Debian
- **Kernel Version:** 6.14.6
- **Product SKU:** L3
- **BIOS Version:** 25.02
- **EC Release:** 3.12

---

### **Logs**
Include any relevant logs that might help diagnose the issue:
- **Dmesg Output:** Attach or paste the relevant portion of `dmesg` logs.
- **Other Logs:** Specify and include logs from related subsystems if applicable.
[   21.599951] intel_tcc_cooling: Programmable TCC Offset detected
[   21.624630] input: HDA Intel PCH Mic as /devices/pci0000:00/0000:00:1f.3/sound/card0/input12
[   21.624678] input: HDA Intel PCH Headphone as /devices/pci0000:00/0000:00:1f.3/sound/card0/input13
[   21.624735] input: HDA Intel PCH HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/sound/card0/input14
[   21.624784] input: HDA Intel PCH HDMI/DP,pcm=7 as /devices/pci0000:00/0000:00:1f.3/sound/card0/input15
[   21.624830] input: HDA Intel PCH HDMI/DP,pcm=8 as /devices/pci0000:00/0000:00:1f.3/sound/card0/input16

### **Additional Context**
Provide any other information or context that might help in debugging this issue. This could include:
- Previous steps taken to resolve the issue.
- Workarounds attempted, if any.
- Frequency of the issue Always

---

Sean-StarLabs, 2025-05-20:
> With headphones connected to audiojack or internal microphone, same result
> caused by the L3 microphone

Paradox ;)

Did it work before the change to fix the headphone detection? Or AMI?

Sean-StarLabs, 2025-05-20:
> "strong noise"

White noise? The noise itself is too loud?

> By the way the latest two version have no rom file, should one just wait until theyure on LVFS and upgrade via fwupd?

Up to you

soc7, 2025-05-20:
> > "strong noise"
> 
> White noise? The noise itself is too loud?
> 
People  on other side say it's like a construction business lol