StarLabsLtd/firmware #312: Improvement: Add BIOS toggle so that keypresses will not wake system when lid is closed

dchang0, 2025-08-03:
Hi--

When carrying my StarLite Mk. V suspended and in its keyboard cover, if the cover is squeezed too hard, it will flex, and one of the keys in the middle of the keyboard will be depressed, thus waking the system. Usually this happens when the system is inside my messenger bag and the bag is jostled while bicycling. Then, a while later, I take the system out of the bag and it is toasty hot, having been awake the whole time inside my bag.

Would it be possible to add a toggle, settable in BIOS, that prevents keypresses from waking the system while the lid sensor indicates that the cover is closed? 

Obviously, keypresses should be allowed to wake the system when the keyboard cover is open, as this would be one of a few expected ways to wake the system for normal use.

Thanks!

Sean-StarLabs, 2025-08-05:
> Would it be possible ...

Ish - we can't target a specific device from firmware, only disable wake from any USB device. Making it conditional on the lid switch will take some doing though. Might be better to do it from the OS, given it's quite specific?

dchang0, 2025-08-05:
Thanks for the info. 

Hmm. 

Wouldn't the system hardware have to wake fully in order for the OS to then make detections and then decide whether to go back to sleep? That's still better than staying awake the whole time while inside my bag, but the OS hasn't been terribly reliable at going back to sleep, at least in Ubuntu.

Maybe this BIOS feature is an idea for a future model.

Other cheesy but workable ideas for future models: stiffen the keyboard cover so that it doesn't flex as much, or recess the keyboard a bit so that there is some space between it and the tablet's surface.

Sean-StarLabs, 2025-08-06:
> Wouldn't the system hardware have to wake fully in order for the OS to then make detections and then decide whether to go back to sleep?

No - you'd disable it as a wake source so it would never wake

dchang0, 2025-08-13:
Thanks, I will check it out. It never occurred to me to read Arch Wiki!